siemens/ros-sharp #119: Can't import URDF asset with WebSocketNet protocol

dwhit, 2018-09-09:
* [x] I am at the right place and my issue is directly related to ROS#. General technical questions I would post e.g. at [ROS Answers](https://answers.ros.org/) or [Stackoverflow](https://stackoverflow.com). For library-specific questions I would look for help in the corresponding library forums.
* [x] I have throughly read [the Contributing Guideline](Contributing.md) and writing this issue is the right thing to do in my case.

---
## I found a bug! ##
* [x] I am using the latest ROS# version available here on the master branch.
* [x] I am adding all required information, code and data files, screenshots and log files so that you can reproduce the problem.

* My OS is: Windows 10
* My Unity Version is: 2017.4.10f
* My ROS Distribution is: Indigo
* My Build Target Platform is: Standalone

**Here is my bug description:**
Hi, I've noticed a bug where when I try to import Baxter via the URDF Asset importer, the process hangs and fails if I'm using WebSocketNet as my protocol. If I'm using WebSocketSharp, it works fine. 

Perform the following steps reproduce the bug:
On ROS side, start a file server node, a WebSocket server, and load a Baxter into Gazebo. 
Open URDF Asset importer window, enter correct IP address, set protocol to NET, and click 'Read Robot Description'

Observed results:
Process hangs at '4. Resources file received."

Expected results:
Baxter is successfully imported.

Relevant Code:
```
N/A
```

MartinBischoff, 2018-09-10:
Thanks for this report @dwhit !
Please update your Unity Editor to a 2018.x Version before trying out anything else.
See also [README.MD/Platform Support](https://github.com/siemens/ros-sharp).
Version 2018.1 is the first offical (not beta) release  supporting .NET 4.5 [see also here](https://blogs.unity3d.com/2018/03/28/updated-scripting-runtime-in-unity-2018-1-what-does-the-future-hold/).

hassanbot, 2018-09-10:
I wasn't able to do the update Friday, but I've tried it now and everything works fine with 2018.2.7f1. Sorry for the inconvenience!

dwhit, 2018-09-12:
after updating to 2018.2, the issue has been resolved. closing the issue. Thanks!

closed 2018-09-12 by dwhit