siemens/ros-sharp #120: Sending simple messages

solernou, 2018-09-13:
* [ x] I am at the right place and my issue is directly related to ROS#. General technical questions I would post e.g. at [ROS Answers](https://answers.ros.org/) or [Stackoverflow](https://stackoverflow.com). For library-specific questions I would look for help in the corresponding library forums.
* [ x] I have throughly read [the Contributing Guideline](Contributing.md) and writing this issue is the right thing to do in my case.

---
## I have a question! ##

* [ x] I searched the [Wiki](https://github.com/siemens/ros-sharp/wiki), [open](https://github.com/siemens/ros-sharp/issues) and [closed](https://github.com/siemens/ros-sharp/issues?q=is%3Aissue+is%3Aclosed) issues for an answer. I tried my best in finding the answer by myself without success. I believe that the discussion we will have in this issue, and the solutions we might find, will help me, and likely other community members who have a similar problem.

**Here is my question:**
I am trying to send a simple string message from Unity through ROS# on a Windows 10 box and get it into a separate Ubuntu 16.04 machine using ROS Kinetic. On Windows I use the Unity package provided in release 1.2c. I have been able to run the tutorial 2.1, successfully importing the turtlebot into Unity, and the Ubuntu firewall does not report any blockage from the Windows box.

I have tried several variations of the code provided in issue 66:
  https://github.com/siemens/ros-sharp/issues/66
without success. 

I wanted to try the latest commit of the library, but could not compile it; while I can compile 1.2c on Visual Studio 2013, newer commits fail to compile on VS2013 and VS2015. 

The only documentation I could find on sending simple messages is on "How to add new message types", and it does not look compatible with version 1.2c. 

Any help?

MartinBischoff, 2018-09-13:
Hi @solernou !
Please use the latest commit [here](https://github.com/siemens/ros-sharp).

You need not  compile anything in Visual Studio yourself.
Just drag&drop [RosSharp](https://github.com/siemens/ros-sharp/tree/master/Unity3D/Assets) into your asset folder.

You can skip steps 1, 2 and 3 in [How to add new Message Types](https://github.com/siemens/ros-sharp/wiki/Dev_NewMessageTypes). Just replace `Messages.Standard.Float32` by the already exiting message type `Messages.Standard.String`.

If this answers your question, please close this issue.

solernou, 2018-09-17:
Hi @MartinBischoff ,
thanks for your very prompt reply. I did not realise that the binaries were included in every commit. I have been able to run send some messages back and forth once I figured out that obviously the Ubuntu side needs the rosbridge_websocket components running. 

I am happily closing this issue.

closed 2018-09-17 by solernou