siemens/ros-sharp #121: outdated dlls in newest Release 1.3 ? e.g. RosBridgeClient.dll

jogitar, 2018-09-17:
* [ X] I am at the right place and my issue is directly related to ROS#. General technical questions I would post e.g. at [ROS Answers](https://answers.ros.org/) or [Stackoverflow](https://stackoverflow.com). For library-specific questions I would look for help in the corresponding library forums.
* [X ] I have throughly read [the Contributing Guideline](Contributing.md) and writing this issue is the right thing to do in my case.

## I found a bug! ##
* [ X] I am using the latest ROS# version available here on the master branch.
* [ X] I am adding all required information, code and data files, screenshots and log files so that you can reproduce the problem.

* My OS is: <Windows 10>
* My Unity Version is: <2018.2.6>
* My ROS Distribution is: <Kinetic Kame>
* My Build Target Platform is: <Windows 10>

**Here is my bug description:**
dll for the newest ROS# Version 1.3 seems to be outdated. I have 60 errors in the Unity Log e.g.
"Assets/RosSharp/Scripts/RosCommuncation/GetParamServiceProvider.cs(17,41): error CS0234: The type or namespace name `Services' does not exist in the namespace `RosSharp.RosBridgeClient'. Are you missing an assembly reference?"

Perform the following steps reproduce the bug:
Create a new Unity Project and Import the newest package from the Unity Asset store (Version 1.3 https://assetstore.unity.com/packages/tools/physics/ros-ros-unity-communication-package-107085 ) or the newest Sourcecode/release from the Github.

Observed results:
60 script compiler errors (seems to be an outdated dll for RosBridgeClient.dll )

Expected results:
no script errors

Relevant Code:
-

suzannahsmith1, 2018-09-18:
Hi @jogitar, 
your issues are most likely due to not referencing the correct .NET Framework in Unity. Please see issue #108 for details. If this does not fix your issue, please let us know.
Suzannah

jogitar, 2018-09-20:
Hi @suzannahsmith1 ,
thank you for your answer. It was my fault. #108 solved it.

closed 2018-09-20 by jogitar