siemens/ros-sharp #125: New URDF workflow makes building very difficult

dwhit, 2018-09-26:
* [x] I am at the right place and my issue is directly related to ROS#. General technical questions I would post e.g. at [ROS Answers](https://answers.ros.org/) or [Stackoverflow](https://stackoverflow.com). For library-specific questions I would look for help in the corresponding library forums.
* [x] I have throughly read [the Contributing Guideline](Contributing.md) and writing this issue is the right thing to do in my case.

---
## I have a question! ##

* [x] I searched the [Wiki](https://github.com/siemens/ros-sharp/wiki), [open](https://github.com/siemens/ros-sharp/issues) and [closed](https://github.com/siemens/ros-sharp/issues?q=is%3Aissue+is%3Aclosed) issues for an answer. I tried my best in finding the answer by myself without success. I believe that the discussion we will have in this issue, and the solutions we might find, will help me, and likely other community members who have a similar problem.

**Here is my question:**

Hi all, the new URDF workflow looks really cool, it's amazing you can create URDFs in Unity now. But the problem I found is that all the new URDF files require UnityEditor, which makes building any project impossible. I haven't found a simple workaround for de-editorifying the codebase, do you have any suggestions?

suzannahsmith1, 2018-09-26:
Hi @dwhit, 
We're glad you find the new workflow helpful, and thank you for bringing this issue to our attention. We didn't realize that the new framework would prevent users from building a Unity project, since our use cases don't require use to build projects. However, this is definitely a valid concern. We are now working on a solution, which should be available soon. 
Thanks,
Suzannah

shoeffner, 2018-09-26:
To add some information, the problem is that the AssetPostprocessor seems to be part for UnityEditor, and as such is not available in Players:

- `Assets/RosSharp/Scripts/Urdf/MeshProcessing/ColladaAssetPostProcessor.cs(26,46): error CS0246: The type or namespace name `AssetPostprocessor' could not be found. Are you missing an assembly reference?`
- `Assets/RosSharp/Scripts/Urdf/MeshProcessing/StlAssetPostProcessor.cs(25,42): error CS0246: The type or namespace name `AssetPostprocessor' could not be found. Are you missing an assembly reference?`

I also haven't found a proper solution yet.

SuzannahSmith, 2018-09-27:
Hello, we made a patch for this issue with pull request #127. Could you try out the new version and see if it solves your problem? Thanks,
Suzannah

shoeffner, 2018-09-27:
Thank you very much, Suzannah – I am able to build the default scene now, except for a warning (just a heads up):
`Script attached to 'LaserScanSphere' in scene '' is missing or no valid script is attached.
UnityEditor.BuildPlayerWindow:BuildPlayerAndRun()`.

MartinBischoff, 2018-10-08:
Thanks for reporting the warning @shoeffner !
It has been fixed in commit https://github.com/siemens/ros-sharp/commit/ac52c7c60c01c02b1c65236173e7524c1006fe03 .

closed 2018-10-08 by MartinBischoff