siemens/ros-sharp #129: Non-convex MeshCollider with non-kinematic Rigidbody is no longer supported in Unity 5

shoeffner, 2018-10-01:
* [x] I am at the right place and my issue is directly related to ROS#. General technical questions I would post e.g. at [ROS Answers](https://answers.ros.org/) or [Stack Overflow](https://stackoverflow.com). For library-specific questions I would look for help in the corresponding library forums.
* [x] I have thoroughly read [the Contributing Guideline](Contributing.md) and writing this issue is the right thing to do in my case.

## I have a problem! ##
* [x] I am using the latest ROS# version available here on the master branch.
* [x] I am adding all required information, code and data files, screenshots and log files so that you can reproduce the problem.

* My OS is: Mac OS 10.13.6
* My Unity Version is: 2018.2.6f1
* My ROS Distribution is: kinetic (via docker container, see [here for dockerfile and entrypoints](https://gitlab.informatik.uni-bremen.de/hoeffner/semantic-simulation/blob/master/ros))
* My Build Target Platform is: mac/linux, but this issue is inside the editor on Mac

**Here is my bug description:**
After importing the PR2 robot when running the scene from the editor, I get the following error:
```
Non-convex MeshCollider with non-kinematic Rigidbody is no longer supported in Unity 5.
If you want to use a non-convex mesh either make the Rigidbody kinematic or remove the Rigidbody component. Scene hierarchy path "pr2/base_footprint/base_link/torso_lift_link/l_shoulder_pan_link/l_shoulder_lift_link/l_upper_arm_roll_link/l_upper_arm_link/l_elbow_flex_link/l_forearm_roll_link/l_forearm_link/l_wrist_flex_link/l_wrist_roll_link/l_gripper_palm_link/l_gripper_r_finger_link/l_gripper_r_finger_tip_link/Collisions/unnamed/l_finger_tip/l_finger_tip_0", Mesh asset path "Assets/Urdf/PR2/pr2_description/meshes/gripper_v0/l_finger_tip_0.asset" Mesh name "l_finger_tip_0"
```
This does not only happen for the finger tip, but for multiple different links.

Perform the following steps reproduce the bug:

1. Clone the repository and run the docker containers to have a working ros instance which publishes the PR2 on the param server. Alternatively, use whatever ROS you have available and publish a PR2 description from [pr2_common](https://github.com/PR2/pr2_common/tree/kinetic-devel/pr2_description) via rosbridge.
```
git clone git@gitlab.informatik.uni-bremen.de:hoeffner/semantic-simulation
cd semantic-simulation
docker-compose up
```
2. Build the master version of the RosSharp.unitypackage using this Makefile (adjust the Unity path on top according to your setup):
```Makefile
UNITY=/Applications/Unity/Unity.app/Contents/MacOS/Unity

UNITY_BATCH=${UNITY} -batchmode -quit -projectPath
BUILD=$(shell pwd)/build
ROS_SHARP_DIR=${BUILD}/ros-sharp

${BUILD}/RosSharp.unitypackage: ${ROS_SHARP_DIR}
	for LIB in RosBridgeClient Urdf; do \
		cd ${ROS_SHARP_DIR}/Libraries && msbuild ${ROS_SHARP_DIR}/Libraries/$${LIB}/$${LIB}.csproj /property:Configuration=Release; \
		cp ${ROS_SHARP_DIR}/Libraries/$${LIB}/bin/Release/* ${ROS_SHARP_DIR}/Unity3D/Assets/RosSharp/Plugins; \
	done
	for EXTERNAL in $(shell ls ${ROS_SHARP_DIR}/Libraries/packages); do \
		cp ${ROS_SHARP_DIR}/Libraries/packages/$${EXTERNAL}/lib/* ${ROS_SHARP_DIR}/Unity3D/Assets/RosSharp/Plugins; \
	done
	# Run Unity once to create project settings -- this tries to compile the
	# scripts which might fail due to the scripting version, which will then be
	# changed.
	-${UNITY_BATCH} ${ROS_SHARP_DIR}/Unity3D
	sed -i '' 's/  scriptingRuntimeVersion: 0/  scriptingRuntimeVersion: 1/g' ${ROS_SHARP_DIR}/Unity3D/ProjectSettings/ProjectSettings.asset
	# Export package
	${UNITY_BATCH} ${ROS_SHARP_DIR}/Unity3D -exportPackage Assets/RosSharp $@ -logFile

${ROS_SHARP_DIR}: | ${BUILD}
	cd ${BUILD} && git clone https://github.com/siemens/ros-sharp.git
	nuget restore -PackagesDirectory ${ROS_SHARP_DIR}/Libraries/packages ${ROS_SHARP_DIR}/Libraries/RosBridgeClient/RosBridgeClient.csproj
	nuget restore -PackagesDirectory ${ROS_SHARP_DIR}/Libraries/packages ${ROS_SHARP_DIR}/Libraries/Urdf/Urdf.csproj

${BUILD}:
	mkdir -p $@
```
3. Create a Unity project, import the RosSharp.unitypackage, set the Player ScriptingRuntimeVersion, restart unity.
4. Use RosBridgeClient to import the PR2. I also added a plane so it doesn't fall down directly (thanks to gravity it wills till fall through though, but slower).
5. Click the run button.

Observed results:
The error messages as depicted above.

When setting all colliders to convex, I receive the following error messages:
```
Invalid worldAABB. Object is too large or too far away from the origin.

Invalid AABB a

Assertion failed on expression: 'IsFinite(d)'

Assertion failed on expression: 'IsFinite(outDistanceForSort)'

Assertion failed on expression: 'IsFinite(outDistanceAlongView)'

Invalid AABB aabb
```

When setting the RigidBodies to Kinematic, I receive the following error:

```
Invalid parameter pos in Rigidbody.WritePose because it was infinity or NaN
```

Expected results:
No error messages.

Additional Information:
I also tried to fix it by adding Twist Subscribers for the link transforms. Currently, what is the recommended way to add multiple subscribers and "fix" the components at once? The manager component used in older versions seems no longer to exist, and I didn't find a proper alternative yet.

MartinBischoff, 2018-10-01:
Hi @shoeffner , 

as the error message said: Unity / PhysX can not work with non-convex mesh colliders. Further info [here](https://docs.unity3d.com/Manual/class-MeshCollider.html).

As you found out by yourself:
(1) setting the mesh colliders convex (and thus using the convex hull of the mesh) works for some meshes.
(2) setting the corresponding rigidbody kinematic (and thus deactivating physics) will make sure you get rid of any further errors in this direction.
(3) another alternative would be to change the URDF in ROS and/or Unity and use a proper simpler collider instead.

In case you want to run a physical simulation of your robot model in Unity, (2) will obviously be no option for you. In case you want to run just an animation of your robot and e.g. subscribe to the corresponding ROS joint_states, (2) should be your preferred option.
Adding Subscribers will not help to get rid of this error.

In case you are asking for the Urdf Patchers in the question you asked in additional information: We moved all those functions to the UrdfRobot script. You find it in the URDFroot GameObject after importing your robot in Unity. Screenshot:
[image]

If this answers your question, please close this issue.

Best,
Martin

shoeffner, 2018-10-01:
Thank you very much, Martin. I didn't see that new script.

If I use it to set convex colliders and disable everything else, I receive the following errors:

```
Invalid worldAABB. Object is too large or too far away from the origin.

Invalid AABB a

Assertion failed on expression: 'IsFinite(d)'

Assertion failed on expression: 'IsFinite(outDistanceForSort)'

Assertion failed on expression: 'IsFinite(outDistanceAlongView)'

Invalid AABB aabb
```

This is unexpected, as the robot is completely contained inside a 2x2x2 box around the origin.
Gravity and Inertia from URDF don't seem to have any influence on this, only setting all rigidbodies to kinematic solves the issue. 

I guess I will try and see if option (2) works for me, otherwise I will have to look into option (3).

Thank you very much!

closed 2018-10-01 by shoeffner

MartinBischoff, 2018-10-02:
Invalid AABB (bounding box) can have various reasons. Google finds multiple long forum discussions.
I remember I had the same error in at least one URDF and went for option (3).
It might be worthwile to try out the suggestions in the [MeshCollider Documentation, Limitations Section](https://docs.unity3d.com/Manual/class-MeshCollider.html), especially enabling __Read/Write__ in the model tab.

If you find out some solution to get rid of this error, and with wich we possibly can improve the (collider) mesh import in ROS#, please let us know.

shoeffner, 2018-10-05:
I found one bounding box which seemed to cause the problems above and removed/disabled it (base_link -> Collisions -> unnamed -> base_L), but that's probably not a general solution. 

[image]

It seems that **Read/Write** is already enabled for the base:

[image]

Unfortunately, now the gravity seems to be pulling on all hinge joints, resulting in a posture making the PR2 look very sad:

[image]

Thus, instead of disabling the collider, I tried using a script on the robot's GameObject to disable collisions between its own joints:

```
	void Start () {
		foreach (Collider c1 in GetComponentsInChildren<Collider>())
			foreach (Collider c2 in GetComponentsInChildren<Collider>())
				Physics.IgnoreCollision(c1, c2, true);
	}
```

This results in a slightly better posture, but I assume that actually collisions with the robot itself (with  custom colliders) might be important to fight gravity:

[image]

However, as the PR2 is a very complex model, I didn't want to create custom colliders for each joint yet, so I decided to follow the URDF tutorial [ROS#: Robot Modeling & Export](https://www.youtube.com/watch?v=07q2mGxD2j8) and create my own robot to test my hypothesis:

[image]

The torso has a weight of 100 to not topple over from the arm on top, which is connected via a hinge joint limited from 0 to 45 degrees (45 degrees being the starting point).

If I enable everything (kinematics, gravity, URDF inertia, convex colliders), of course the robot is completely still when running the simulation. The same is if I disable kinematics, and gravity. But if I disable kinematics and URDF inertia, the arm falls down. Going back to the PR, if I disable URDF inertia, its body parts just quickly fly out of scope just after spawning.

[image]

So I guess ineartia and joints are the key here. I haven't found a proper configuration yet and am wondering if it's simply a case of the URDF data not really being compatible with the physics simulation in Unity. Do you have any experience with this?