siemens/ros-sharp #130: Issuses during subscribing a Custom MsgType

KonradBielecki, 2018-10-04:
## I have a question! ##

Hello,

I have created my own MsgType on the ROS side (see below rqt-Monitor):
_____________topic_____________________________________type
[image]

Now I want to subsrcibe it on the Unity side. For that, I create a own MSG in the RosSharp Asset 1.3, following the instructions of he Wiki:
[image]

I also created a Subscriberscript in Unity:
[image]

The issue is, that I can not subscribe he hole MSG. I am publishing values on the ROS side, but only receive the integer values back. The Vector3 elements are always empty or in this case 0. (see below Unity screenshot);
[image]

I tried to do solve it, like in issue #36 mentioned, but without success. 

Any on that can help, please?? 

Best regards 
Konrad

bsubbaraman, 2018-10-05:
I am also experiencing an issue with subscribing to a custom type, though I am a step behind you.  As a test, I copied this [example](https://github.com/siemens/ros-sharp/wiki/Dev_NewMessageTypes) exactly, copy and pasting the provided code.  However, in the FloatSubscriber script, I get the error "the type or namespace Float32 does not exist in the namespace 'RosSharp.RosBridgeClient.Messages.Standard'.  Did you follow any additional steps outside of the ones provided in that link?  If I get this working, I can hopefully help debug your problem.

KonradBielecki, 2018-10-05:
Hi,
I followed the example like mentioned in the tutorial. There was no problem with it. 
I think your issue, can cause two things: 
1. you didn't include your Float32 Mesasge in the RosSharp.sln project
2. you haven't build the RosBridgeClient.dll, and copied it in the Unity project path of RosSharp

It would be helpful if you post some screenshots of your code and maybe the structure of directories.

Greetings
Konrad

MartinBischoff, 2018-10-05:
@KonradBielecki :
Please make sure that the message you define in RosSharp exactly represents the ROS message type.

Your ROS message has one `geometry_msgs/Twist` message named`twsitCommand`  whereas your C# class `VehicleGuidanceCommands` has 2: `linear` and `angular`.  This will not work. You will have to replace
```
Geometry.Twist linear
Geomerty.Twist angular
```
by
```
Geometry.Twist twistCommand
```

@bsubbaraman : 
* Please open a new issue for new problems in the future. 
 * If your new message type can not be found in your Unity project, make sure that you performed steps 2 and 3 as described in the tutorial. In case you deactived automatic asset import, make sure to click Assets/Reimport All after Step 3 of the tutorial.

@KonradBielecki Please do not forget to close this issue when your problem is solved.