siemens/ros-sharp #131: Issue Adding New Message Type

bsubbaraman, 2018-10-05:
Hello,

I have a question, which I am moving to a separate issue from [issue #130](https://github.com/siemens/ros-sharp/issues/130).  I  am trying to add a new message type- as a test, I am following this [tutorial](https://github.com/siemens/ros-sharp/wiki/Dev_NewMessageTypes).  However, after many, many attempts, I haven't been able to find success.  

I download [ROS# 1.3](https://github.com/siemens/ros-sharp/releases) and in Libraries/RosBridgeClient/Messages/Standard I add the Float32 class, copy and pasted from the tutorial.  I then navigate to Libraries/RosShar.sln and build the solution.  This is the output of that:
[image]

I then open a new Unity project, and import ROS# from the Asset store.  I replace the RosBridgeClient.dll in Unity3D/Assets/RosSharp/Plugins with the new one generated from Libraries/RosBridgeClient/bin/Release.  I also re-import all assets.  I then create a new Float32 subscriber script, copy and pasted from the tutorial, but I receive the error "the type or namespace Float32 does not exist in the namespace 'RosSharp.RosBridgeClient.Messages.Standard'".  Alternatively, I've tried opening a new, empty Unity Project, and dragging and dropping the Unity3D/Assets/RosSharp folder into my project hierarchy directly, but receive the same error.  

Any help on steps I'm overlooking/etc are greatly appreciated.  Is it a problem with building the solution?  Thank you.

closed 2018-10-05 by bsubbaraman

MartinBischoff, 2018-10-08:
Hi @bsubbaraman !
You closed the issue without a comment. Have you solved your problem yourself?
Best,
Martin

bsubbaraman, 2018-10-08:
Yes!  I added my solution as an edit at the top of my question.  The problem was that Visual Studio did not add the new class to the solution.  
Thank you for the help!