siemens/ros-sharp #132: Building the ros-sharp Libraries

solernou, 2018-10-08:
* [x] I am at the right place and my issue is directly related to ROS#. General technical questions I would post e.g. at [ROS Answers](https://answers.ros.org/) or [Stack Overflow](https://stackoverflow.com). For library-specific questions I would look for help in the corresponding library forums.
* [x] I have thoroughly read [the Contributing Guideline](Contributing.md) and writing this issue is the right thing to do in my case.

---

## I have a question! ##

* [x] I searched the [Wiki](https://github.com/siemens/ros-sharp/wiki), [open](https://github.com/siemens/ros-sharp/issues) and [closed](https://github.com/siemens/ros-sharp/issues?q=is%3Aissue+is%3Aclosed) issues for an answer. I tried my best to find the answer by myself without success. I believe that the discussion we will have in this issue, and the solutions we might find, will help me, and likely other community members who have a similar problem.

## I found a bug! ##
* [x] I am using the latest ROS# version available here on the master branch.
* [x] I am adding all required information, code and data files, screenshots and log files so that you can reproduce the problem.

* My OS is: Windows 10
* My Unity Version is: not relevant
* My ROS Distribution is: not relevant
* My Build Target Platform is: Windows 10, any CPU

This is not exactly a bug, but I am not able to build the Libraries, or more precisely RosBridgeClient. I am using MS Visual Studio 15 and the error comes at lines 124 and 125 of Communicators.cs:

            bool isSuccess = ServiceCallHandler.Invoke(args.ToObject<Tin>(), out Tout result);
            return new ServiceResponse<Tout>(id, Service, result, isSuccess);

for which the compiler complains that:
   'Tout' is a type but is used like a variable  // line 124
and then:
  The name 'result' does not exist in the current context  // line 125

That is using the latest commit, 251bceb8, but started earlier. 

The solution provided to build the libraries targets VS2015, but maybe you've been using a newer compiler... any chances for a quick fix?

MartinBischoff, 2018-10-08:
Hi @solernou !

We use  .NET Framework 4.6. This is also the latest .NET framework supported by the latest offical  Unity release.

I presume your compile error comes from [Covariance and Contravariance](https://docs.microsoft.com/en-us/dotnet/csharp/programming-guide/concepts/covariance-contravariance/index) which has been introduced in C# 4.0.

Please upgrade to .NET Framework 4.6.

solernou, 2018-10-08:
Hi @MartinBischoff , 
thanks for the quick reply. I've just double checked the .NET framework, and this doesn't look like the cause of the error. Here,
[image]
you can see that v. 4.6 is installed, and here:
[image]
that it is correctly configured (AFAIK, as I am still a Linux chap), but that I still get the errors:
[image]
when starting from a clean repo. 

Anything else I could try? 

Thanks,
Albert

MartinBischoff, 2018-10-10:
Sorry, I can not reproduce and do not understand these errors. Please find my screenshot of a successful compilation of the most [recent commit](https://github.com/siemens/ros-sharp/commit/251bceb8530a73fa193068b9be47bef75abc0731) below.

I have doubts if the following suggestions solve this problem:

__Try Visual Studio 2017__
Theoretically .NET 4.6 Framework should be identical in every Visual Studio editor supporting it. Nevertheless, since we are using rather recent C# syntax here, some covanriance syntax checks from the older Visual Studio version might sound alarm. 

__Is your repo really 'clean'?__
Esp. I do not understand the second error that you marked on your screenshot. Why is it looking for the dll that should be built? Why is the dll denoted' Metadata file'? Is there any file that has been autogenerated by Visual Studio from a previous version in your repo, that might cause these errors?

[image]

Sorry for not coming up with any better suggestions and for any inconvenience caused. You will definitely find better support on this C# problem on e.g. https://stackoverflow.com/ . Maybe also someone else watching these issues has a better idea?

Please do inform us on any progress you make on this issue.

solernou, 2018-10-12:
Hi, 
thanks for the tip. It took me a while to find some time to install VS2017, but it certainly did the trick. 

Perhaps you'd like to change the solution, so that it reflects it needs v 2017, instead of 2015? 

You can well close this issue. 

Thanks,
Albert

MartinBischoff, 2018-10-12:
I'm happy to read that you got it working with Visual Studio 2017!

Actually the solution file already says that the solution was generated using Visual Studio 2017 (which was given the version number 15).  I found an overview on Visual Studio solution file headers [here](https://gist.github.com/DanAtkinson/3f863464a5dadd93b8e4dea6fe7b973a).

We'll update the [README.md](https://github.com/siemens/ros-sharp/blob/master/README.md) to reduce the risk of further misunderstandings.

closed 2018-10-12 by MartinBischoff