siemens/ros-sharp #133: Import of .dae-embedded materials fails during URDF import

shoeffner, 2018-10-08:
* [x] I am at the right place and my issue is directly related to ROS#. General technical questions I would post e.g. at [ROS Answers](https://answers.ros.org/) or [Stack Overflow](https://stackoverflow.com). For library-specific questions I would look for help in the corresponding library forums.
* [x] I have thoroughly read [the Contributing Guideline](Contributing.md) and writing this issue is the right thing to do in my case.

---
## I found a bug! ##
* [x] I am using the latest ROS# version available here on the master branch.
* [x] I am adding all required information, code and data files, screenshots and log files so that you can reproduce the problem.

* My OS is: MacOS 10.14
* My Unity Version is: 2018.2.11f1
* My ROS Distribution is: kinetic
* My Build Target Platform is: mac/linux

**Here is my bug description:**
Some .dae files, e.g. pr2's [should_pan.dae](https://github.com/PR2/pr2_common/blob/kinetic-devel/pr2_description/meshes/shoulder_v0/shoulder_pan.dae) use custom, embedded materials, which are not loaded when importing the URDF from ROS or from file.

Perform the following steps reproduce the bug:
- You can use the attached unitypackage file (it's a "pdf" so github allows to upload it... just change the name to actually use it) [PR2_urdf.unitypackage.pdf](https://github.com/siemens/ros-sharp/files/2456290/PR2_urdf.unitypackage.pdf), or use your ROS# setup as explained in #129 to import the pr2.
- Import the robot_description.urdf using the context menu import function.

Observed results:
[image]

Expected results:
Instead of the blue and green meshes, they should use their proper materials. 

For example, the Assets/Urdf/PR2/pr2_description/meshes/shoulder_v0/shoulder_pan.dae can be dragged into the scene and has proper textures then, as unity recognizes them properly:

[image]

This is different from #110 as the texturization is not happening via embedded textures but embedded materials.

MartinBischoff, 2018-10-10:
Thank you @shoeffner for this thorough report. It was very easy to understand and to reproduce.

_"It's not a bug, it's a feature!"_ 😏 

Seriously: The original dae textures are overridden by the material definition in the URDF.

If you want to have the original dae material textures, just delete the corresponding lines in the urdf file.
For example, in the definition for the head visuals, lines 926-932:
```
    <visual>
      <origin rpy="0.0 0.0 0.0" xyz="0 0 0"/>
      <geometry>
        <mesh filename="package://pr2_description/meshes/head_v0/head_tilt.dae"/>
      </geometry>
      <material name="Green"/>
    </visual>
```
If you delete `<material name="Green"/>` you will have the original dae texture instead:
[image]

I find this override order (URDF over dae) more practical and user friendly than the other way round:
* Individual meshes (dae files) are normally used in other programs (e.g. CAD software) where users might want to apply other textures than in the URDF robot system model.
* If users make the effort to additionally define a material in the URDF, this definition should win.

Please let me know in case you have any remarks or comments on this. If not, please don't forget to close the issue.

shoeffner, 2018-10-10:
Thank you, I guess I will change the URDF then!
Both cases make sense to me, I understood the materials inside the URDF more like a backup, but this is a perfect explanation. Much appreciated!

closed 2018-10-10 by shoeffner

dwhit, 2018-10-18:
Hey, I know this issue is closed, but RViz defaults the opposite way, i.e. the DAE material takes precedence over the URDF color. I understand if you don't want to change it, but where in the code would I change it so it matches RViz?

dwhit, 2018-10-18:
Nevermind, I found it in UrdfMaterial.cs