siemens/ros-sharp #137: Unity model explodes

kiyukuta, 2018-10-25:
* [x] I am at the right place and my issue is directly related to ROS#. General technical questions I would post e.g. at [ROS Answers](https://answers.ros.org/) or [Stack Overflow](https://stackoverflow.com). For library-specific questions I would look for help in the corresponding library forums.
* [x] I have thoroughly read [the Contributing Guideline](Contributing.md) and writing this issue is the right thing to do in my case.

---
## I have a question! ##

* [x] I searched the [Wiki](https://github.com/siemens/ros-sharp/wiki), [open](https://github.com/siemens/ros-sharp/issues) and [closed](https://github.com/siemens/ros-sharp/issues?q=is%3Aissue+is%3Aclosed) issues for an answer. I tried my best to find the answer by myself without success. I believe that the discussion we will have in this issue, and the solutions we might find, will help me, and likely other community members who have a similar problem.

**Here is my question:**

I followed the turtlebot's tutorial with unity_simulation_scene.launch in ROS side.
When I play the scene the 3D model in Unity explodes.
Do I miss some setting?

[image]

MartinBischoff, 2018-10-26:
Hi @kiyukuta !

Though spectacular, an exploding robot is not what we want to see at this point...
Obviously you have missed something from the video, but we can only guess what.
Models normally explode like this if  colliders intersect such that the physics engine can not resolve collisions properly.

For debugging you can compare the reference scene UnitySimulationScene in the RosSharp/Scenes folder with the scene that you set up based on the tutorial. Though the meshes are missing in the reference scene, it is fully functional and you can control an invisible turtlebot that does not explode. You can see the turtlebot's colliders and joints by selecting the turtlebot GameObject in the scene view. You can also see it moving like this when the application is running.
Please compare the GameObjects with Rigidbody components and see if the rigidbody options and the joint options are identical to that of the reference scene. The GameObjects wheel_left_link and wheel_right_link should have only the following two additional components: JointStateReader and JoyAxisMotorWriter.

Is there anyone who ran into a similar problem when going through the tutorials and can advise here?

Please do not forget to report here and to close the issue when you found the solution.

kiyukuta, 2018-11-02:
Hi @MartinBischoff ! Sorry for late reply.

I noticed that it happens when `Use URDF Data` in `Urdf Inertial` component is false.
This can be set to true by clicking `Enable` button of  `Use Inertia from URDF` in `UrdfRobot` Component.

The tutorial video only enables `Use gravity` and `Convex`, so I pressed disable button of others. That was the reason.
https://youtu.be/sup2Tdxt1ko?t=125

Now my turtlebot works fine. Thanks!

closed 2018-11-02 by kiyukuta

HelloWorldLTY, 2022-04-10:
Hi, did you successfully run this program in this part? We did the same settings as yours but failed...... Could you please help us? Thanks.