siemens/ros-sharp #138: Adding message types

dwhit, 2018-11-08:
* [x] I am at the right place and my issue is directly related to ROS#. General technical questions I would post e.g. at [ROS Answers](https://answers.ros.org/) or [Stack Overflow](https://stackoverflow.com). For library-specific questions I would look for help in the corresponding library forums.
* [x] I have thoroughly read [the Contributing Guideline](Contributing.md) and writing this issue is the right thing to do in my case.

## I have a feature request! ##
Would people be interested in adding a bunch of message type?

For my research, I've added a lot of messages that aren't provided by default in ROS#. visualization_msgs, moveit_msgs, nav_msgs, as well as some geometry, sensor, and other ones not included. Would it be of interest to pull request them into the master branch?

MartinBischoff, 2018-11-09:
Thanks @dwhit for this early pull request warning.
Feel free to make a pull request for those message types of which you think they will used by many and are worth adding, downloading and adminsitrating.

Rhybo, 2019-01-29:
Has there been any update on this pull request?  Personally, I am trying to include some mavros messages (specifically [NavSatFix](http://docs.ros.org/api/sensor_msgs/html/msg/NavSatFix.html) and [NavSatStatus](http://docs.ros.org/api/sensor_msgs/html/msg/NavSatStatus.html)) in my project.

MartinBischoff, 2019-03-04:
Hi @dwhit , how shall we proceed with this issue?
Are you planning to make a pull request?
Please check out commit https://github.com/siemens/ros-sharp/commit/b5b1db955a72d284711aa456f2f110e97fb12b1f with which we can now comfortably add new messages in Unity via a dialoge window.
What do you think about a ROS# Feature Store where we can advertise your fork with your new message types?

dwhit, 2019-03-06:
The feature store sounds like a good idea! and the new message generation looks really nice.

closed 2019-05-31 by berkayalpcakal