siemens/ros-sharp #140: Video Using Ros-Sharp

ybchoo, 2018-11-20:
* [ ] I am at the right place and my issue is directly related to ROS#. General technical questions I would post e.g. at [ROS Answers](https://answers.ros.org/) or [Stack Overflow](https://stackoverflow.com). For library-specific questions I would look for help in the corresponding library forums.
* [ ] I have thoroughly read [the Contributing Guideline](Contributing.md) and writing this issue is the right thing to do in my case.

---
## I have a question! ##

* [ ] I searched the [Wiki](https://github.com/siemens/ros-sharp/wiki), [open](https://github.com/siemens/ros-sharp/issues) and [closed](https://github.com/siemens/ros-sharp/issues?q=is%3Aissue+is%3Aclosed) issues for an answer. I tried my best to find the answer by myself without success. I believe that the discussion we will have in this issue, and the solutions we might find, will help me, and likely other community members who have a similar problem.

**Here is my question:**

---
## I found a bug! ##
* [ ] I am using the latest ROS# version available here on the master branch.
* [ ] I am adding all required information, code and data files, screenshots and log files so that you can reproduce the problem.

* My OS is: 
* My Unity Version is: 

* My ROS Distribution is: 
Kinetic
* My Build Target Platform is: 

**Here is my bug description:**

Perform the following steps reproduce the bug:

Observed results:

Expected results:

Relevant Code:
```
<!-- Insert code here. Optionally attach code or data files to this issue. -->
```
---
## I have a feature request! ##

**I want to do this:**

**For being able to do it, I wish ROS# had the following feature:**

ybchoo, 2018-11-20:
I have tried to work on Video Streaming using ImageSubscriber but observed that video capture only the first image while all other images were not display. Can experts in Ros Sharp kindly help?
          
public void takeVideo()
{
  this.instance.imageSubscription_id = rosSocket.Subscribe<sensor_msgs.Image>(
               "/camera/color/image_raw", subscribeVideoHandler);
}

public void subscribeVideoHandler(sensor_msgs.Image image)
{
// we can retrieve first video image from here but subsequent video images were not capture.
}
Can someone kindly advises?

MartinBischoff, 2018-11-26:
Hi @ybchoo !
From the code you shared in this issue, I see no reason, why only the first image should be received.
Your error most likely depends on how you use the two methods and the class containing it.
* Please make sure that you call `takeVideo()`only once to subscribe the topic.
* Please make sure that the corresponding class instance stays alive such that the corresponding callback can be called.

Please check out [UnitySimulationScene](https://github.com/siemens/ros-sharp/blob/master/Unity3D/Assets/RosSharp/Scenes/UnitySimulationScene.unity), the corresponding [Wiki page](https://github.com/siemens/ros-sharp/wiki/User_App_ROS_UnitySimulationExample) and [YouTube Video](https://youtu.be/sup2Tdxt1ko).
This tutorial also explains how to subscribe image data from ROS, with the only difference, that we're using `CompressedImage` message types there...

ybchoo, 2018-11-27:
Hi Dr Martin
We tried CompressedImage.
On RosBridge end, it shows as attached.