siemens/ros-sharp #141: How do I import Gazebo world.file(Not a URDF) in to Unity?

stevensu1838, 2018-11-22:
Hi all,

I am trying to import the following robot and its environment into Unity. 
[image]

However, I've noticed in this ROS package. Only the robot hand and robot are in URDF file. The whole environment is a world.file of Gazebo. 

```
model.urdf
model_hand.urdf
smart_grasping_mod.world
```

Anyone knows how to import a world.file of Gazebo into Unity? If not, is it doable that I convert a world file into URDF then import it into Unity? Thanks.

stevensu1838, 2018-11-22:
That's part of the world file.
[image]

roehrlverena, 2018-11-22:
Hello,
to convert a SDF to a URDF file, you might want to take a look at [this repo](https://github.com/andreasBihlmaier/pysdf). 
A tutorial on how to use it with a robot model can be found [here](https://www.youtube.com/watch?v=8g5nMxhi_Pw).

Best,
Verena

MartinBischoff, 2018-11-22:
I'm not aware of a world / SDF -> Unity importer.
In fact we discussed about adding one to ROS# in this issue: #35 
We decided against it, as this file format is very Gazebo-specific.
So, in accordance to @roehrlverena 's answer above, I'd also suggest to get URDF files out of the `<model>` sections in the world/SDF file first and then use the exisiting URDF importer.

Is there data, in the SDF that you would like to have imported in Unity, beside that in the `<model>` tags?
Please do keep us informed on the progress you make.

MartinBischoff, 2018-12-12:
Hi @stevensu1838 any update on your sdf model import?
How did you proceed?

MartinBischoff, 2019-01-07:
closing due to inactivity

closed 2019-01-07 by MartinBischoff