siemens/ros-sharp #144: Map Visualization in Unity

merci194, 2018-12-03:
* [x] I am at the right place and my issue is directly related to ROS#. General technical questions I would post e.g. at [ROS Answers](https://answers.ros.org/) or [Stack Overflow](https://stackoverflow.com). For library-specific questions I would look for help in the corresponding library forums.
* [x] I have thoroughly read [the Contributing Guideline](Contributing.md) and writing this issue is the right thing to do in my case.

---
## I have a question! ##

* [x] I searched the [Wiki](https://github.com/siemens/ros-sharp/wiki), [open](https://github.com/siemens/ros-sharp/issues) and [closed](https://github.com/siemens/ros-sharp/issues?q=is%3Aissue+is%3Aclosed) issues for an answer. I tried my best to find the answer by myself without success. I believe that the discussion we will have in this issue, and the solutions we might find, will help me, and likely other community members who have a similar problem.

**Here is my question:**
 I'm trying to use the Occupancy grid message definition to visualize the 2D map i have made on Unity.As a reference I'm using the Laser Scan Subscriber and Visualizer scripts. These scripts on their own helped me understand a lot about how meshes in Unity are represented so thank you for that.(I'm a Unity Beginner) However i feel a little lost trying to depict the map in Unity. 

For starters i dont understand how the visualiser would have to be implemented. What would be a good approach? 

Should i try to create a maze with the Occupancy grid values? where occupied areas are walls and free areas are passages? 

 How do i go about aligning the origins of the map in ros to the one in Unity ,since unity uses a different coordinate system?Are there any specific coordinate transforms i have to take care of?

Or is it better to try a totally different approach such as trying to save and load a binary map?

Is there an implementation of the map visualisation already available or one you are working on?

**I want to do this:**
 I want to map with SLAM , localise and navigate a turtle bot on the map on Unity.During the course of my project i intend to implement Pure Localisation and Move Base visualisation on Unity as a sort of GUI frontend to ROS. I want to create a nice user interface for the turtlebot.The ROS# project has been really helpful in importing URDF files and visualising the Laser Scanner data. 

**For being able to do it, I wish ROS# had the following feature:**
  Something like the visualiser scripts written for the Laser Scanner with the origins of both maps(in ROS and Unity) being aligned is what i'm looking for.It would really help someone new to Unity who'l looking to create graphical frontend applications to get started,especially since mesh generation can be a little time consuming to get into.I think it might also help rapid prototyping in industrial environments where people can sometimes be averse to using the terminal interface of Linux :)

MartinBischoff, 2018-12-05:
Hi @merci194 ,

I'd suggest to model the occupancy grid data as a Unity [Mesh](https://docs.unity3d.com/ScriptReference/Mesh.html). You can write your own Occupancy grid visualizer in an identical way as the laser scanner example.
This is a beautiful training task for Unity beginners 😏 

I shy away of adding further visualiziation example as for example for the an occupancy grid to the ROS# master, as it is very application-specific in terms of update rate, positioning, size and orientation, performance, coloring, shading, etc.

Please feel free to use this issue for exchanging Occupancy Grid visualization methods. Maybe @awesome-manuel @MischaRo @dwhit  have something in their pocket...

MartinBischoff, 2018-12-12:
Hi @merci194 any update on your Occupancy grid visualizer?
How shall we proceed with this issue?

MartinBischoff, 2019-01-07:
closing due to inactivity

closed 2019-01-07 by MartinBischoff

Caveats, 2020-04-30:
@merci194 Did you ever resolve this?