siemens/ros-sharp #146: Missing ServiceConsumer class？

EckoTan0804, 2018-12-04:
* [x] I am at the right place and my issue is directly related to ROS#. General technical questions I would post e.g. at [ROS Answers](https://answers.ros.org/) or [Stack Overflow](https://stackoverflow.com). For library-specific questions I would look for help in the corresponding library forums.
* [x] I have thoroughly read [the Contributing Guideline](Contributing.md) and writing this issue is the right thing to do in my case.

---
## I have a question! ##

* [x] I searched the [Wiki](https://github.com/siemens/ros-sharp/wiki), [open](https://github.com/siemens/ros-sharp/issues) and [closed](https://github.com/siemens/ros-sharp/issues?q=is%3Aissue+is%3Aclosed) issues for an answer. I tried my best to find the answer by myself without success. I believe that the discussion we will have in this issue, and the solutions we might find, will help me, and likely other community members who have a similar problem.

**Here is my question:**

1. In **ros-sharp/Libraries/RosBridgeClient/Communicator.cs** I have found classes named `ServiceProvider` and `ServiceConsumer`. But in **ros-sharp/Unity3D/Assets/RosSharp/Scripts/RosBridgeClient/RosCommunication/** I have only found the `ServiceProvider.cs` and its implementation `GetParamServiceProvider.cs`. Is the class `ServiceConsumer` missing?

2. I have read the code of `ServiceProvider` but I am not pretty sure it i have understood it correctly. 

   ~~~C#
   using UnityEngine;
   
   namespace RosSharp.RosBridgeClient
   {
       [RequireComponent(typeof(RosConnector))]
       public abstract class ServiceProvider<Tin, Tout> : MonoBehaviour where Tin : Message where Tout : Message
       {
           public string ServiceName;
   
           protected virtual void Start()
           {
               GetComponent<RosConnector>().RosSocket.AdvertiseService<Tin, Tout>(ServiceName, ServiceCallHandler);
           }
   
           protected abstract bool ServiceCallHandler(Tin request, out Tout response);
   
       }
   }
   ~~~

   1. Should I use the method `ServiceCallHandler` to handle the request and respone of my service? If I didnt't get wrong, the parameter `request` is the message I want to send to ROS and the parameter `response` is the response message from ROS. If I want to do something with the response, I just need to write my handling of the `response` in this method. Am I right?
   2. When will `ServiceCallHandler` be called?

MartinBischoff, 2018-12-05:
Hi @EckoTan0804 ,

this is quite a bunch of questions...

1. Topics are typically published and subscribed on `Start()`, and services are provided on `Start()`, but services are generally not called on `Start()`, but any time on demand during simulation. In contrast to `Publisher`, `Subscriber` and `ServiceProvider` I see no use of having a general `ServiceConsumer` class in Unity (what should it generalize?). Instead I suggest to directly call the Service when needed in Unity, e.g. via:
```
 RosSocket.CallService<rosapi.GetParamRequest, rosapi.GetParamResponse>("/rosapi/get_param", ServiceCallHandler, new rosapi.GetParamRequest("/rosdistro", "default"));
```
If you have a suggestion for a useful ServiceConsumer parent class, please share.

2. short answers:
i. yes! yes!
ii. yes!
longer answer: Please check out [GetParamServiceProvider](https://github.com/siemens/ros-sharp/blob/master/Unity3D/Assets/RosSharp/Scripts/RosBridgeClient/RosCommuncation/GetParamServiceProvider.cs)  for an implementation example of the abstract class `ServiceProvider` that you cited. See also issue #101 .

argo422, 2018-12-07:
Hi @MartinBischoff 
Thanks for your answer. That helps me a lot!

MartinBischoff, 2018-12-12:
@argo422 @EckoTan0804 Happy to her that. I consider this issue closed.
Nevertheless, in case you find any improvements on the class structure please let me know.

closed 2018-12-12 by MartinBischoff