siemens/ros-sharp #147: Change topic /camera/rgb/image_raw/compressed  to /camera/rgb/image_raw

stevensu1838, 2018-12-05:
Hi Sir,
ROS sharp release 1.3
It worked well before when I subscribed topic /camera/rgb/image_raw/compressed
Purpose: change the image on image receiver to a colour one
Topics to be subscribed by Unity from ROS using ROS#:
Change this topic /camera/rgb/image_raw/compressed (**sensor_msgs/CompressedImage**) to /camera/rgb/image_raw(**sensor_msgs/Image**)

I've noticed that both sensor_msgs/CompressedImage and sensor_msgs/Image are built in message type. So I don't think I need to create my own type of message.

What I did: change all **SensorCompressedImage** to **SensorImage** in order to receive topic /camera/rgb/image_raw

```
using System;
using UnityEngine;

namespace RosSharp.RosBridgeClient
{
    [RequireComponent(typeof(MeshRenderer))]
    public class ImageReceiver : MessageReceiver
    {
        public override Type MessageType { get { return (typeof(SensorImage)); } }

        private byte[] imageData;
        private bool isMessageReceived;

        private MeshRenderer meshRenderer;
        private Texture2D texture2D;

        private void Awake()
        {
            MessageReception += ReceiveMessage;
        }
        private void Start()
        {
            texture2D = new Texture2D(1,1);
            meshRenderer = GetComponent<MeshRenderer>();
            meshRenderer.material = new Material(Shader.Find("Standard"));
        }
        private void Update()
        {
            if (isMessageReceived)
                ProcessMessage();
        }
        private void ReceiveMessage(object sender, MessageEventArgs e)
        {
            imageData = ((SensorImage)e.Message).data;
            isMessageReceived = true;
        }

        private void ProcessMessage()
        {
            texture2D.LoadImage(imageData);
            texture2D.Apply();
            meshRenderer.material.SetTexture("_MainTex", texture2D);
            isMessageReceived = false;

```

Then I rebuilt the image receiver in Unity by following the video tutorial. In rqt_graph it shows ROS is publishing **/camera/rgb/image_raw** through ROSbridge successfully. However, I don't get any image on the plane. **(It worked before when I subscribed topic /camera/rgb/image_raw/compressed )**
[image]
Can you please kindly walk me through? Thanks a lot

MartinBischoff, 2018-12-05:
Most likely the method [Texture2D.LoadImage()](https://docs.unity3d.com/530/Documentation/ScriptReference/Texture2D.LoadImage.html) does not work with the data contained in [sensor_msgs/Image](http://docs.ros.org/api/sensor_msgs/html/msg/Image.html).
`sensor_msgs/Image`contains bitmap data whereas `LoadImage()` works with png and jpg.
You will have to find a way to import bitmap files into Unity.

MartinBischoff, 2018-12-12:
Hi @stevensu1838 ! Any update on the problem you were having?
Please give a short update and close the issue is your problem is solved.

MartinBischoff, 2019-01-07:
closing due to inactivity

closed 2019-01-07 by MartinBischoff