siemens/ros-sharp #149: can not import urdf to unity, ubuntu system

Xunzhaocunzi, 2018-12-06:
* [ ] I am at the right place and my issue is directly related to ROS#. General technical questions I would post e.g. at [ROS Answers](https://answers.ros.org/) or [Stack Overflow](https://stackoverflow.com). For library-specific questions I would look for help in the corresponding library forums.
* [ ] I have thoroughly read [the Contributing Guideline](Contributing.md) and writing this issue is the right thing to do in my case.

---
I have been spent a few days trying to make it work. Could someone help? Thank you.

I stuck at importing urdf file from ros to unity, which is step 2.1

I ran "roslaunch file_server publish_description_turtlebot2.launch", and everything looks good. 
--------
qiwei@qiwei-desktop:~$ roslaunch file_server publish_description_turtlebot2.launch 
... logging to /home/qiwei/.ros/log/2e5a49fe-f97d-11e8-a527-7085c2a3653b/roslaunch-qiwei-desktop-16514.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

WARN: unrecognized 'param' tag in <include> tag
WARN: unrecognized 'param' tag in <include> tag
xacro: Traditional processing is deprecated. Switch to --inorder processing!
To check for compatibility of your document, use option --check-order.
For more infos, see http://wiki.ros.org/xacro#Processing_Order
xacro.py is deprecated; please use xacro instead
started roslaunch server http://qiwei-desktop:38273/

SUMMARY
========

PARAMETERS
 * /robot/name: Turtlebot2
 * /robot_description: <?xml version="1....
 * /rosapi/params_glob: [*]
 * /rosapi/services_glob: [*]
 * /rosapi/topics_glob: [*]
 * /rosbridge_websocket/address: 
 * /rosbridge_websocket/authenticate: False
 * /rosbridge_websocket/bson_only_mode: False
 * /rosbridge_websocket/delay_between_messages: 0
 * /rosbridge_websocket/fragment_timeout: 600
 * /rosbridge_websocket/max_message_size: None
 * /rosbridge_websocket/params_glob: [*]
 * /rosbridge_websocket/port: 9090
 * /rosbridge_websocket/retry_startup_delay: 5
 * /rosbridge_websocket/services_glob: [*]
 * /rosbridge_websocket/topics_glob: [*]
 * /rosbridge_websocket/unregister_timeout: 10
 * /rosdistro: kinetic
 * /rosversion: 1.12.14

NODES
  /
    file_server (file_server/file_server)
    rosapi (rosapi/rosapi_node)
    rosbridge_websocket (rosbridge_server/rosbridge_websocket)

auto-starting new master
process[master]: started with pid [16527]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 2e5a49fe-f97d-11e8-a527-7085c2a3653b
process[rosout-1]: started with pid [16540]
started core service [/rosout]
process[rosbridge_websocket-2]: started with pid [16557]
process[rosapi-3]: started with pid [16558]
process[file_server-4]: started with pid [16559]
[ INFO] [1544117668.776074406]: ROSbridge File Server initialized.
registered capabilities (classes):
 - rosbridge_library.capabilities.call_service.CallService
 - rosbridge_library.capabilities.advertise.Advertise
 - rosbridge_library.capabilities.publish.Publish
 - rosbridge_library.capabilities.subscribe.Subscribe
 - <class 'rosbridge_library.capabilities.defragmentation.Defragment'>
 - rosbridge_library.capabilities.advertise_service.AdvertiseService
 - rosbridge_library.capabilities.service_response.ServiceResponse
 - rosbridge_library.capabilities.unadvertise_service.UnadvertiseService
[INFO] [1544117669.104394]: Rosbridge WebSocket server started on port 9090
[INFO] [1544117700.555239]: Client connected.  1 clients total.
[INFO] [1544117700.637328]: Client disconnected. 0 clients total.
[INFO] [1544117705.788375]: Client connected.  1 clients total.
[INFO] [1544117705.839788]: Client disconnected. 0 clients total.
[INFO] [1544117706.487450]: Client connected.  1 clients total.
[INFO] [1544117706.524153]: Client disconnected. 0 clients total.
---------------------------------
However, I stuck here: 
[image]

* [ ] I searched the [Wiki](https://github.com/siemens/ros-sharp/wiki), [open](https://github.com/siemens/ros-sharp/issues) and [closed](https://github.com/siemens/ros-sharp/issues?q=is%3Aissue+is%3Aclosed) issues for an answer. I tried my best to find the answer by myself without success. I believe that the discussion we will have in this issue, and the solutions we might find, will help me, and likely other community members who have a similar problem.

**Here is my question:**

---
## I found a bug! ##
* [ ] I am using the latest ROS# version available here on the master branch.
* [ ] I am adding all required information, code and data files, screenshots and log files so that you can reproduce the problem.

* My OS is: 
* My Unity Version is: 
* My ROS Distribution is: 
* My Build Target Platform is: 

**Here is my bug description:**

Perform the following steps reproduce the bug:

Observed results:

Expected results:

Relevant Code:
```
<!-- Insert code here. Optionally attach code or data files to this issue. -->
```
---
## I have a feature request! ##

**I want to do this:**

**For being able to do it, I wish ROS# had the following feature:**

MartinBischoff, 2018-12-07:
Hi @Xunzhaocunzi ,
in the UrdfAssetImporterDialog change the addess to that of your Ubunutu VM.
`localhost` won't work. You are not setting up the websocket connection to the system you are connecting from.

Enter `ìfconfig`in a ubuntu console. The IP you need is the one of interface  `ènp0s8`.
Please read the [Wiki](https://github.com/siemens/ros-sharp/wiki/User_Inst_UbuntuOnOracleVM).

Please close the issue when this solves your problem.

Xunzhaocunzi, 2018-12-07:
> Hi @Xunzhaocunzi ,
> in the UrdfAssetImporterDialog change the addess to that of your Ubunutu VM.
> `localhost` won't work. You are not setting up the websocket connection to the system you are [...]

I am not running VM, instead, both unity3d and ROS are running in the ubuntu 16.04 system. 
You mentioned some people were able to run ros# purely in Linux system. Would you have a plan to update the tool for Linux in the near future? Anyway, thank you for developing and sharing the tool with us.
[image]

MartinBischoff, 2018-12-12:
Hi @Xunzhaocunzi 
ROS# has also been used on Linux systems since more than one year. Please check out the issues.
https://github.com/siemens/ros-sharp/issues/29 where we solved the last platform-specific bugs 

Websocket connection requires two different machines.
Please feel free to extend ROS# to your needs. You will have to implement a new [IProtocol](https://github.com/siemens/ros-sharp/blob/master/Libraries/RosBridgeClient/Protocols/IProtocol.cs) interface implementation that comunicates directly (e.g. via shared memory) instead of using a Websocket connection.
Best,
Martin

closed 2018-12-12 by MartinBischoff