siemens/ros-sharp #151: What is the exact command to install gazebo7.11?

Xunzhaocunzi, 2018-12-08:
* [ ] I am at the right place and my issue is directly related to ROS#. General technical questions I would post e.g. at [ROS Answers](https://answers.ros.org/) or [Stack Overflow](https://stackoverflow.com). For library-specific questions I would look for help in the corresponding library forums.
* [ ] I have thoroughly read [the Contributing Guideline](Contributing.md) and writing this issue is the right thing to do in my case.

---
What is the exact command to install gazebo 7.11 please? I could not open gazebo from roslaunch, and I did search online and could not find a line of code that was working. Thanks a lot.

* [ ] I searched the [Wiki](https://github.com/siemens/ros-sharp/wiki), [open](https://github.com/siemens/ros-sharp/issues) and [closed](https://github.com/siemens/ros-sharp/issues?q=is%3Aissue+is%3Aclosed) issues for an answer. I tried my best to find the answer by myself without success. I believe that the discussion we will have in this issue, and the solutions we might find, will help me, and likely other community members who have a similar problem.

DanielArnett, 2019-01-25:
Here's a hint @Xunzhaocunzi 

I removed gazebo

`sudo apt-get remove gazebo7*`

Added Open Source Robotics to my sources

`sudo sh -c 'echo "deb http://packages.osrfoundation.org/gazebo/ubuntu-stable `lsb_release -cs` main" > /etc/apt/sources.list.d/gazebo-stable.list'`

Added their key.

`wget http://packages.osrfoundation.org/gazebo.key -O - | sudo apt-key add -`

`sudo apt-get update`

Installed

`sudo apt-get install gazebo7`

Then I had 7.14.

berkayalpcakal, 2019-01-26:
The main reason why we suggest users to have Gazebo with such particular version is because we once experienced an issue that Gazebo did not even work with Ubuntu on the VirtualBox. So if you can run Gazebo alone on your native Ubuntu, then there should not be any problem with the version of Gazebo.

I believe that the problem you are having is irrelevant to the topic of the issue here.