siemens/ros-sharp #153: Error with Pointcloud.cs script

akashcastelino, 2018-12-17:
Hi,
I have encountered an issue with the Pointcloud.cs script.
The pointcloud constructor(taking subscribed depth and rgb images) gives a Null Reference Error when used to create a PointCloud object. Is anyone else facing this issue?

@MartinBischoff in your comment in a different post, you've mentioned that pointcloud.cs script gets the points in 3D from the 2 2D images. Could you please clarify?
Obtaining the points in 3D from a PointCloud2 message works( but as expected is slow), but the other constructor that takes 2 2D images to give 3D position gives the above-mentioned error.

The line in **bold** below is where the error is first happens.

public PointCloud(Image depthImage, Image rgbImage, float focal)
        {

            int width = depthImage.width;
            int height = depthImage.height;
            float invFocal = 1.0f / focal;

            Points = new RgbPoint3[width * height];

            for (int v = 0; v < height; v++)
            {
                for (int u = 0; u < width; u++)
                {
                    float depth = 0;// depthImage[u, v];
                    if (depth == 0)
                    {
                        **Points[u + v * width].x = float.NaN;**
                        Points[u + v * width].y = float.NaN;
                        Points[u + v * width].z = float.NaN;
                        Points[u + v * width].rgb = new int[] { 0, 0, 0 };
                    }
                    else
                    {
                        Points[u + v * width].z = depth * invFocal;
                        Points[u + v * width].x = u * depth * invFocal;
                        Points[u + v * width].y = v * depth * invFocal;
                        Points[u + v * width].rgb = new int[] { 0, 0, 0 };// rgbImage[u, v];
                    }
                }
            }
        }

Also since there seems to be many who are trying to visualise pointcloud data in unity with Ros-Sharp, I request you to kindly consider sharing your turtlebot application code somewhere outside the Ros-Sharp repository, as it would not need to be administrated or enhance readability outside ros sharp and at the same time would offer a lot of help with visualising point cloud data.

* My OS is: Windows 10
* My Unity Version is: 2018.2
* My ROS Distribution is: Kinetic

akashcastelino, 2019-01-10:
Hi @MartinBischoff , Is there any update on this issue ?

MartinBischoff, 2019-01-10:
Dear @akashcastelino ,

it seems like there is no update. If I had helpful answers to the questions in your mail, I would have answered them already.

Maybe it would help to address your questions not to me personally?
In the end this is a public issue board in an open source project and not a one-person-helpdesk.

Maybe also a more precise formulation of the actual problem you are having would help?

My _little or not helpful_ comments to the questions you wrote above are:

(1) I have not yet faced the issue you described above when creating a PointCloud object with the PointCloud.cs class.
(2) As discussed in the other post, we created a mesh of 3d points out of 2 2d images, an rgb image and a depth image. It is exactly the method you referenced above which I was talking about.
(3) Thank you also for appeciating the turtlebot application code. I am aware that people are interested in our stuff and we are taking your request into consideration. Please let myself decide where and when I upload which of my source code.

Sorry that I can not help futher for the time being.

Best,
Martin

closed 2019-06-14 by berkayalpcakal