siemens/ros-sharp #154: Unity crashes when breaking ros bridge communication

nlamprian, 2019-01-04:
* My OS is: Windows 10 (and Ubuntu 16.04)
* My Unity Version is: 2018.3.0f2
* My ROS Distribution is: Kinetic
* My Build Target Platform is: Windows

**Here is my bug description:**
When playing the unit simulation example, if rosbridge communication is broken, after a while Unity crashes by a message "Fatal Error in GC. Unexpected mark stack overflow".

Perform the following steps reproduce the bug:
1. Launch unity_simulation_scene in ROS.
2. Start simulation in Unity.
3. Kill ROS (Ctrl+C in the terminal).
4. Wait for the window with the message to appear.

Observed results:
Unity crashes.

MartinBischoff, 2019-01-07:
Thank you for this issue @nlamprian .
In the `UnitySimulationScene` the `RosConnector` component uses the "Web Socket NET" protocol by default.
Does this bug also happen on your system after switching to the "Web Socket Sharp" protocol?

nlamprian, 2019-01-07:
With "Web Socket Sharp", simulation terminates cleanly.
I tested only on Linux this time. Let me know if you want me to check on Windows.

MartinBischoff, 2019-01-07:
Thanks for checking @nlamprian , then we have isolated this porblem to [WebSocketNetProtocol.cs](https://github.com/siemens/ros-sharp/blob/master/Libraries/RosBridgeClient/Protocols/WebSocketNetProtocol.cs).

[WebSocketNetProtocol.cs](https://github.com/siemens/ros-sharp/blob/master/Libraries/RosBridgeClient/Protocols/WebSocketNetProtocol.cs) is a very miminalistic WebSocket Client implementation based on the [System.NET WebSockets](https://docs.microsoft.com/de-de/dotnet/api/system.net.websockets?view=netframework-4.7.2) Namespace.

At first glance it seems to me like we do not check whether the server is still active before starting to send or to listen. If the server is not active, this stack overflow error is caused on the client side.

sseiichi, 2020-05-01:
It's a bother for me to check if the ROS server exists every time.
I want ROS# to check if the server exists before connection...

MartinBischoff, 2020-05-05:
We improved [WebSocketNetProtocol](https://github.com/siemens/ros-sharp/blob/master/Libraries/RosBridgeClient/Protocols/WebSocketNetProtocol.cs) in the meantime 
 and can no longer reproduce the error.

@nlamprian Please let me know if this issue persists on your side with the latest ROS# version.

closed 2020-05-05 by MartinBischoff