siemens/ros-sharp #155: Received a message with wrong topic

secondage, 2019-01-06:
In Unity3D, When i have two topics, like topic1 and topic12, i use two subscribers to subscribe them, When topic1 receives the message, topic12 will also receive the same message callback.

on RosShape.cs->SubscribersOf() return the subscribers list startwith "topic1", so the "topic12"  is also filtered into the list, What is the purpose of doing this?

thank you

MartinBischoff, 2019-01-07:
Thank you for this comment @secondage .
Using `StartsWith()` in [RosSocket](https://github.com/siemens/ros-sharp/blob/master/Libraries/RosBridgeClient/RosSocket.cs), Method `SubscribersOf()`, Line 177, is not correct.
It will give the wrong results as you highlighted in the example above.

Would `Equals()` work for you instead?

secondage, 2019-01-07:
I suggest changing to `StartWith(topic + ":")` because ":" has been added to the topic name in the `GetUnusedCounterID` method, so the use of `StartWith(topic + ":")`should avoid most errors.

MartinBischoff, 2019-01-07:
done in commit https://github.com/siemens/ros-sharp/commit/33b641df76da2e0535bd90efb354658bb3d8d086.

closed 2019-01-07 by MartinBischoff