siemens/ros-sharp #157: ArgumentNullException when closing RosSocket

Piflik, 2019-01-09:
I am using ROS# in a current project to display a live feed from a camera on a drone and I regularly get an ArgumentNullException when I close the RosSocket. I don't use one of the RosConnector MonoBehaviours, but I create the connection in my own custom class. However, when I open the "UnitySimulationScene" that is part of the unitypackage and connect to the same topic with a RosConnector MonoBehaviour I get the same Exception when I exit Play-Mode. The Connector I use is a modified ImageConnector that I changed in order to receive an Image-Message as opposed to a CompressedImage-Message. 

In my own class I open and close the socket during runtime. 

The exception does **not** affect the functionality of my program. Everything continues to work. The error also does not show up every time I close the socket.

Sometimes I get an ObjectDisposedException instead, or a SocketException stating that the connection has been closed by the remote host. I could not reproduce these with the RosConnector, but the first exception during the runtime of my program is _always_ the ArgumentNullException. These two exceptions only occur on subsequent calls to RosSocket.Close() without exiting Play-Mode. 

The error appears both in Editor and in the build.

**OS:** Windows 10 64bit
**Unity Version:** 2018.3.0f2
**ROS# Release:** Version 1.4
**Target Platform:** Standalone Windows and Android (later also iOS)

Stack Trace (emphasis mine): 

ArgumentNullException: Value cannot be null.
Parameter name: src
System.Buffer.BlockCopy (System.Array src, System.Int32 srcOffset, System.Array dst, System.Int32 dstOffset, System.Int32 count) (at <ac823e2bb42b41bda67924a45a0173c3>:0)
System.Net.WebSockets.ManagedWebSocket+<ReceiveAsyncPrivate>d__60.MoveNext () (at <3845a180c26b4889bc2d47593a665814>:0)
Rethrow as OperationCanceledException: Aborted
System.Net.WebSockets.ManagedWebSocket+<ReceiveAsyncPrivate>d__60.MoveNext () (at <3845a180c26b4889bc2d47593a665814>:0)
--- End of stack trace from previous location where exception was thrown ---
System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw () (at <ac823e2bb42b41bda67924a45a0173c3>:0)
System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess (System.Threading.Tasks.Task task) (at <ac823e2bb42b41bda67924a45a0173c3>:0)
System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification (System.Threading.Tasks.Task task) (at <ac823e2bb42b41bda67924a45a0173c3>:0)
System.Runtime.CompilerServices.TaskAwaiter.ValidateEnd (System.Threading.Tasks.Task task) (at <ac823e2bb42b41bda67924a45a0173c3>:0)
System.Runtime.CompilerServices.ConfiguredTaskAwaitable`1+ConfiguredTaskAwaiter[TResult].GetResult () (at <ac823e2bb42b41bda67924a45a0173c3>:0)
System.Net.WebSockets.ManagedWebSocket+<CloseAsyncPrivate>d__66.MoveNext () (at <3845a180c26b4889bc2d47593a665814>:0)
--- End of stack trace from previous location where exception was thrown ---
System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw () (at <ac823e2bb42b41bda67924a45a0173c3>:0)
System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess (System.Threading.Tasks.Task task) (at <ac823e2bb42b41bda67924a45a0173c3>:0)
System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification (System.Threading.Tasks.Task task) (at <ac823e2bb42b41bda67924a45a0173c3>:0)
System.Runtime.CompilerServices.TaskAwaiter.ValidateEnd (System.Threading.Tasks.Task task) (at <ac823e2bb42b41bda67924a45a0173c3>:0)
System.Runtime.CompilerServices.TaskAwaiter.GetResult () (at <ac823e2bb42b41bda67924a45a0173c3>:0)
**RosSharp.RosBridgeClient.Protocols.WebSocketNetProtocol+<Close>d__20.MoveNext () (at <cb8ef69c0a9548e3b5c8fbf26c4f61fc>:0)**
--- End of stack trace from previous location where exception was thrown ---
System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw () (at <ac823e2bb42b41bda67924a45a0173c3>:0)
System.Runtime.CompilerServices.AsyncMethodBuilderCore+<>c.<ThrowAsync>b__6_0 (System.Object state) (at <ac823e2bb42b41bda67924a45a0173c3>:0)
UnityEngine.UnitySynchronizationContext+WorkRequest.Invoke () (at C:/buildslave/unity/build/Runtime/Export/UnitySynchronizationContext.cs:111)
UnityEngine.UnitySynchronizationContext.Exec () (at C:/buildslave/unity/build/Runtime/Export/UnitySynchronizationContext.cs:72)
UnityEngine.UnitySynchronizationContext.ExecuteTasks () (at C:/buildslave/unity/build/Runtime/Export/UnitySynchronizationContext.cs:93)

[Stack Trace - ArgumentNullException.txt](https://github.com/siemens/ros-sharp/files/2741000/Stack.Trace.-.ArgumentNullException.txt)
[Stack Trace - ObjectDisposedException.txt](https://github.com/siemens/ros-sharp/files/2741001/Stack.Trace.-.ObjectDisposedException.txt)
[Stack Trace - SocketException.txt](https://github.com/siemens/ros-sharp/files/2741002/Stack.Trace.-.SocketException.txt)

Piflik, 2019-01-10:
Judging from RosSocket.cs, RosSocket.Close() already unadvertizes/unsubscribes from all topics that have been advertized or subscribed to.
I have tried explicitly unsubscribing from the topic before closing, but that didn't change the behaviour. 

I don't publish anything from Unity and only listen to the one topic for the image data. The Unity application might connect/disconnect multiple times (and might even shut down completely in between) while the other end of the connection keeps running (and communicates with a third application) and doesn't even know if Unity is connected or not, except for the additional subscriber.

This is the Connector I reproduced it with:

	using UnityEngine;

	namespace RosSharp.RosBridgeClient {
		[RequireComponent(typeof(RosConnector))]
		public class ImageSubscriber : Subscriber<Messages.Sensor.Image> {
			public MeshRenderer meshRenderer;

			private Texture2D texture2D;
			private byte[] imageData;
			private bool isMessageReceived;

			private int width, height;

			protected override void Start() {
				base.Start();
				texture2D = new Texture2D(1, 1);
				meshRenderer.material = new Material(Shader.Find("Standard"));
			}
			private void Update() {
				if (isMessageReceived)
					ProcessMessage();
			}

			protected override void ReceiveMessage(Messages.Sensor.Image compressedImage) {
				imageData = compressedImage.data;
				width = compressedImage.width;
				height = compressedImage.height;
				isMessageReceived = true;
			}

			private void ProcessMessage() {
				if (texture2D == null || texture2D.width != width || texture2D.height != height) {
					texture2D = new Texture2D(width, height, TextureFormat.RGB24, false);

					meshRenderer.material.SetTexture("_MainTex", texture2D);
				}
				texture2D.LoadRawTextureData(imageData);
				texture2D.Apply();
				isMessageReceived = false;
			}

		}
	}

I can only assume that the receiver thread still tries to process a message that has been sent before the shutdown was received from the host, but arrived after the Unity-Client has closed its end of the connection. I tried adding an early return in the callback I subscribe to the topic with, but that didn't help, either. Neither did delaying the RosSocket.Close() call by a frame after unsubscribing.

Here are the relevant excerpts of the code I am using:

	private void TryConnectROS() {
		if (string.IsNullOrEmpty(_rosTopic) || string.IsNullOrEmpty(_ip) || !IPAddress.TryParse(_ip, out _)) return;

		if (_rosSocket == null || !_rosSocket.protocol.IsAlive()) {
			CloseROS(); //probably unnecessary
			WebSocketNetProtocol protocol = new WebSocketNetProtocol($"ws://{_ip}:{_port}");

			protocol.OnConnected += (sender, args) => {
				if (string.IsNullOrEmpty(_rosSubscription)) {
					_rosSubscription = _rosCompressed
						? _rosSocket.Subscribe<RosSharp.RosBridgeClient.Messages.Sensor.CompressedImage>(_rosTopic, ReceiveROSPackage, 1)
						: _rosSocket.Subscribe<RosSharp.RosBridgeClient.Messages.Sensor.Image>(_rosTopic, ReceiveROSPackage, 1);
				}
			};

			_rosSocket = new RosSocket(protocol);
		}
	}

	private void CloseROS() {
		if (string.IsNullOrEmpty(_rosSubscription)) return;
		_rosSocket?.Unsubscribe(_rosSubscription);
		CoroutineRunner.Delay(() => _rosSocket?.Close());
	}

	private void ReceiveROSPackage(RosSharp.RosBridgeClient.Messages.Sensor.Image imageData) {
		if (!_rosSocket.protocol.IsAlive()) return;
		_textureWidth = imageData.width;
		_textureHeight = imageData.height;
		_textureBytes = imageData.data;

		_textureDirty = true;
	}

	private void ReceiveROSPackage(RosSharp.RosBridgeClient.Messages.Sensor.CompressedImage imageData) {
		if (!_rosSocket.protocol.IsAlive()) return;
		_textureBytes = imageData.data;

		_textureDirty = true;
	}

Piflik, 2019-01-15:
I have found a 'solution'. I replaced the dll with the source code and replaced the ClientWebSocket.CloseAsync() call with ClientWebSocket.CloseOutputAsync(). I don't know enough about WebSockets, but I haven't got the above errors since. The other end of the connection still reports a successfully closed connection, so I might be fine.

I noticed that CloseAsnyc oftentimes never completed. It might be that the Server didn't accept the close request. But as I said, I don't know enough about WebSockets, so I might also just be masking the problem.

MartinBischoff, 2019-01-15:
Hi @Piflik ,

thank you for this info. [Here in Section 3 (Teardown)](http://www.salmanq.com/blog/5-things-you-probably-didnt-know-about-net-websockets/2013/04/) is a nice explanation on the differences between `CloseOutputAsync` and `CloseAsync`.
`CloseAsync` is the polite way, it informs the partner before pulling the plug. But this send operation might interfere with currently ongoing send operations.

To close the connection _politely_ we will have to change the behaviour as follows:
* immediately stop sending any new messages
* wait until all ongoing send operations have finished
* then send the `CloseAsync`message

Have you tried switching to `websocket-sharp` protocol in the RosConnector component? Is that an option for you?
`WebSocketNetProtocol` is a WebSocket Client implementation that we wrote as an alternative to `websocket-sharp` which does not support newer frameworks as .NET Standard 2.0. The teardown procedure in `WebSocketNetProtocol`  has room for improvement  as we also found out in  #154 .

Piflik, 2019-01-16:
I tried the WebSocketSharpProtocol, but when I use it the video playback is extremely slow with uncompressed videos. They work perfectly fine with the WebSocketNetProtocol (as long as the bandwidth to the server is good enough). 

For testing I use a Virtual Machine running Ubuntu and I get these framerate issues here already. We had to switch to compressed video for the actual drone, because the bandwidth wasn't adequate for uncompressed, so I fear that when I switch to websocket-sharp even compressed video might not work with the hardware, but I'll have to test this.

MartinBischoff, 2019-01-16:
Interesting to read that the WebSocketNetProtocol is faster in your case.
Issue #66 is about the performance of WebSocketNetProtocol. There might still be room for improvement.

berkayalpcakal, 2020-03-26:
Closed due to inactivity.

closed 2020-03-26 by berkayalpcakal