siemens/ros-sharp #158: How To Send --once?

ybchoo, 2019-01-11:
* [ ] I am at the right place and my issue is directly related to ROS#. General technical questions I would post e.g. at [ROS Answers](https://answers.ros.org/) or [Stack Overflow](https://stackoverflow.com). For library-specific questions I would look for help in the corresponding library forums.
* [ ] I have thoroughly read [the Contributing Guideline](Contributing.md) and writing this issue is the right thing to do in my case.

---
## I have a question! ##

* [ ] I searched the [Wiki](https://github.com/siemens/ros-sharp/wiki), [open](https://github.com/siemens/ros-sharp/issues) and [closed](https://github.com/siemens/ros-sharp/issues?q=is%3Aissue+is%3Aclosed) issues for an answer. I tried my best to find the answer by myself without success. I believe that the discussion we will have in this issue, and the solutions we might find, will help me, and likely other community members who have a similar problem.

**Here is my question:**

Hi RosSharp Gurus
When I execute 
rostopic pub /topicname std_msgs/Empty --once
it works and --once is mandatory. Without --once, ros command will not work.

My Json object 
JSON Object  {"topic":"/topicname","type":"std_msgs/Empty","op":"advertise","id":"/topicname"}

{
  "topic": "/topicname",
  "msg": 
  {
    "data": " --once"
  },
  "op": "publish",
  "id": "/topicname"
}

Can kindly advises on how can we send "--once" from RosSharp to RosBridge without 
param name data embedded in msg?
Thanks n look forward hearing from RosSharp.

## I found a bug! ##
* [ ] I am using the latest ROS# version available here on the master branch.
* [ ] I am adding all required information, code and data files, screenshots and log files so that you can reproduce the problem.

* My OS is: 
* My Unity Version is: 
* My ROS Distribution is: 
* My Build Target Platform is: 

**Here is my bug description:**

Perform the following steps reproduce the bug:

Observed results:

Expected results:

Relevant Code:
```
<!-- Insert code here. Optionally attach code or data files to this issue. -->
```
---
## I have a feature request! ##

**I want to do this:**

**For being able to do it, I wish ROS# had the following feature:**

MartinBischoff, 2019-01-11:
[RosbridgeClient](https://github.com/siemens/ros-sharp/tree/master/Libraries/RosBridgeClient) implements the [rosbridge protocol](https://github.com/RobotWebTools/rosbridge_suite/blob/develop/ROSBRIDGE_PROTOCOL.md).
In Section 3.4.3 of that protocol the format of a publish message is specified:
```
{ "op": "publish",
  (optional) "id": <string>,
  "topic": <string>,
  "msg": <json>
}
```
there is no additional JSON field like `param` that we could use to communicate the `-once` or other parameters to ROS.

It is a helpful info however to read how you managed communicate the `-once` param in the msg field.

Two remarks:
* You could suggest exending the publish message format on the [rosbridge_suite](https://github.com/RobotWebTools/rosbridge_suite) issue board for future rosbridge_suite versions.

* Have you thought about implementing your topic as a service or action instead?
This is often the better solution than a ``-once` param topic.

ybchoo, 2019-01-11:
*>> your topic as a service or action instead *
Thanks Dr Martin for the advises. I will work on this
*>> rosbridge_suite
<https://github.com/RobotWebTools/rosbridge_suite> issue board .*
I have already posted it, thanks for the suggestion.
Really appreciated your prompt action and time. Great work, I luv it.

closed 2019-01-14 by MartinBischoff