siemens/ros-sharp #159: ros-sharp can transfer octomap_msgs

victor123499, 2019-01-15:
i want to ask some question
can this tool transfer the octoamp_msgs to unity
i want to transfer my octomap to unity
does any body know how to do?

MartinBischoff, 2019-01-15:
Hi @victor123499 ,
you can find all ROS message types which are natively supported by ROS# in [RosBridgeClient/Messages](https://github.com/siemens/ros-sharp/tree/master/Libraries/RosBridgeClient/Messages).
[Ocotomap](https://github.com/OctoMap/octomap_msgs/blob/melodic-devel/msg/Octomap.msg) is currently not among the natively supported message types.
But you can add it yourself as described in the [Wiki page](https://github.com/siemens/ros-sharp/wiki/Dev_NewMessageTypes).
Please close this issue if it answers your question.

victor123499, 2019-01-15:
Thanks @MartinBischoff 
do you know where can teach me how to add the octomap_msgs
i know can add it be myself
but i'm not really know how to use

MartinBischoff, 2019-02-27:
Dear @victor123499 sorry I do not understand the comment above. The [wiki page](https://github.com/siemens/ros-sharp/wiki/Dev_NewMessageTypes) (also linked above) provides a detailed explanation how to add new message types.

When you recevie the data in Unity, you can decide by youeself what you want to do with it. Please look at the various examples in the demo scenes.

closed 2019-02-27 by MartinBischoff