siemens/ros-sharp #161: Problems when sending messages with arrays

solernou, 2019-01-30:
* [v] I am at the right place and my issue is directly related to ROS#. General technical questions I would post e.g. at [ROS Answers](https://answers.ros.org/) or [Stack Overflow](https://stackoverflow.com). For library-specific questions I would look for help in the corresponding library forums.
* [v] I have thoroughly read [the Contributing Guideline](Contributing.md) and writing this issue is the right thing to do in my case.

## I found a bug! ##
* [x] I am using the latest ROS# version available here on the master branch. 
   * I was using ROS# release 1.4.
* [v] I am adding all required information, code and data files, screenshots and log files so that you can reproduce the problem.

* My OS is: Windows 10
* My Unity Version is: 2018.2, 2018.3
* My ROS Distribution is: Kinetic
* My Build Target Platform is: Windows Desktop.

**Here is my bug description:**
When there is an array in my message, the receiver does not get the right results. To illustrate this, I am sending a Sensor::CompressedImage at every fixed update, using the header.seq to mark the order of the messages, and writing on the array modulo 8.

Perform the following steps reproduce the bug:
 Get RosSharp v 1.4, open the unity package in Unity. Add a RosConnector, it does not matter whether WebSocket Net or Sharp. Add the following publisher:

    using UnityEngine;
    namespace RosSharp.RosBridgeClient
    {
        public class MyByteArrayPublisher : Publisher<Messages.Sensor.CompressedImage>
        {
            int cnt;
            protected override void Start()
            {
                base.Start();
                cnt = 0;
            }

            private void FixedUpdate()
            {
                Messages.Sensor.CompressedImage m = new Messages.Sensor.CompressedImage();
                m.header.seq = cnt;
                m.data = new byte[400 * 400];
                m.data[0] = (byte)(cnt % 8);
                m.data[2] = (byte)(cnt % 8);
                cnt += 1;
                Debug.Log("M: " + m.header.seq + ", " + m.data[2]);
                Publish(m);
            }
        }
    }

On the ROS side, I have the following C++ receiver:

    #include "ros/ros.h"
    #include "sensor_msgs/CompressedImage.h"

    void imageCallback(const sensor_msgs::CompressedImage::ConstPtr& msg)
    {
      ROS_INFO("I heard: [%7d %8u %8u %8u %8u %8u]", msg->header.seq, msg->data[0],
               msg->data[1], msg->data[2], msg->data[3], msg->data[4]);  
    }

    int main(int argc, char **argv)
    {
      ros::init(argc, argv, "listener");
      ros::NodeHandle n;
      ros::Subscriber sub = n.subscribe("unity_bytes", 1000, imageCallback);
      ros::spin();

      return 0;
    }

**Observed results:**
One can see in the picture, that the Unity Editor prints messages correctly, but my C++ listener in Ubuntu do not. While the messages are received in order (header.seq looks fine), one can see that data starts well, but quickly gets is not what should be.
[image]

**Expected results:**
Obviously, data should be the same on both Unity Editor and C++ listener. Running the following C++ publisher works as expected.

    #include "ros/ros.h"
    #include "sensor_msgs/CompressedImage.h"

    int main(int argc, char **argv)
    {
      ros::init(argc, argv, "speakingBytes");
      ros::NodeHandle n;
      ros::Publisher chatter_pub = n.advertise<sensor_msgs::CompressedImage>("unity_bytes", 1000);
      ros::Rate loop_rate(10);
      int count = 0;
      while (ros::ok())
      {
        sensor_msgs::CompressedImage imgMsg;

        imgMsg.header.seq = count;
        imgMsg.data = std::vector<uint8_t>(400*400, 0);
        imgMsg.data[2] = (uint8_t) (count % 8);

        ROS_INFO("I said: %7d %8u", imgMsg.header.seq, imgMsg.data[2]);

        chatter_pub.publish(imgMsg);
        ros::spinOnce();
        loop_rate.sleep();
       ++count;
      }

      return 0;
    }

solernou, 2019-02-12:
Anybody?

MartinBischoff, 2019-02-12:
Hi @solernou !
* When the message size exceeds a certain limit it is split up into multiple pieces. If additionally to many of those messages must be sent in too short time, more recent messages are re-composed before older ones on the other side and the order is lost. I also remeber that some messages were lost in these cases, most likely due to buffer/memory limits. 

* I presume that this change in order happens on the server / ROS side. At least the WebSocketNet Client that I implemented can in no case change the order of messages. Another indication for that is, that both WebSocketSharp and WebSocketNet yield the same problem.

* I remember that I simply reduced the traffic load by reducing the time step between messages.

* Another option would be to ask [here](https://github.com/RobotWebTools/rosbridge_suite) for support.

* Other than that, why do you publish an image every `FixedUpdate()` anyway? Unity renders an image every `Update()` only. By publishing every `FixedUpdate()` you will often send the same image multiple times.  If you publish in `LateUpdate()` , you will have the most freshly rendered data available 
and at the same time drastically reduce your network load...
To further reduce load, you could implement a counter which only publishes every 2nd or 3rd `LateUpdate()` or so....

I hope this helps.

MartinBischoff, 2019-02-27:
Hi @solernou did my answer help? Any update on your issue?

solernou, 2019-02-27:
Hi Martin,
yes, your answer was really clarifying indeed. We're investigating 
whether we can live with less data traffic, but haven't been able to 
work on it as have been busy with some other projects.

I'll write you back when I have something useful :)

Cheers,
Albert

solernou, 2019-03-14:
After some investigation I managed to make it work at 12Hz under the Unity Editor. In order to do this benchmark I set the frequency on both sides to it, while keeping the FixedUpdate on the Unity side. I also switched to LG's implementation of RosBridge this time. We're having internal discussions on whether to accept this frequency or not.

Something else I checked was to set the header ID to a fixed value:
      imgMsg.header.seq = 1;
and to my surprise, this did not have an effect on the receptor side. Instead, the received messages showed the expected behaviour (1, 2, 3, 4, ...). Thus, it is either that:
 - ROS# fixes the header if it is invalid.
 - RosBridge does not care of this value and prints whatever it likes. In that case, we could be seeing out-of-order messages when the workload is too heavy for RosBridge to cope with. 

I am happy for you to close this issue if you don't have anything further to add.

MartinBischoff, 2019-03-14:
Thank you for reporting these results @solernou !

`imgMsg.header` will be corrected by ROS#, only if you call 'imgMsg.header.Update()'.
[The corresponding method is here.](https://github.com/siemens/ros-sharp/blob/master/Unity3D/Assets/RosSharp/Scripts/Extensions/StandardHeaderExtensions.cs).

I also see no open point here for the ROS# side and close this issue.

closed 2019-03-14 by MartinBischoff