siemens/ros-sharp #163: ROS Sharp in Melodic

hojjatabdollahi, 2019-02-18:
* [v ] I am at the right place and my issue is directly related to ROS#. General technical questions I would post e.g. at [ROS Answers](https://answers.ros.org/) or [Stack Overflow](https://stackoverflow.com). For library-specific questions I would look for help in the corresponding library forums.
* [v ] I have thoroughly read [the Contributing Guideline](Contributing.md) and writing this issue is the right thing to do in my case.

---
## I have a question! ##

* [v ] I searched the [Wiki](https://github.com/siemens/ros-sharp/wiki), [open](https://github.com/siemens/ros-sharp/issues) and [closed](https://github.com/siemens/ros-sharp/issues?q=is%3Aissue+is%3Aclosed) issues for an answer. I tried my best to find the answer by myself without success. I believe that the discussion we will have in this issue, and the solutions we might find, will help me, and likely other community members who have a similar problem.

**Here is my question:**
I searched everywhere and I couldn't find the answer to this question:
Does Ros Sharp support Melodic? Does it need a specific version of ROS Bridge?

I have been using Ros Sharp (in Unity) in Kinetic successfully for the past year. But now I am moving my code to Ubuntu 18.04 and ROS Melodic, but unfortunately, my code does not has stopped working. I can see the terminal output that "1 client has connected" but the next message that says "[client 0] Subscribed to /mytopic" is not showing up and I suspect it is because the protocol has changed from Kinetic to Melodic and ROS sharp is using the previous protocol (maybe).

MartinBischoff, 2019-02-19:
@hojjatabdollahi , we only need not check whether [rosbridge_suite](http://wiki.ros.org/rosbridge_suite) works with ROS melodic and that the interface, i.e. [rosbridge protocol V.2](https://github.com/RobotWebTools/rosbridge_suite/blob/develop/ROSBRIDGE_PROTOCOL.md) did not change.

I have not tried ROS# with ROS melodic yet, but at first glance on the [ROS Wiki](http://wiki.ros.org/rosbridge_suite) it looks like this package is fully implemented for melodicand all commands and specifications are identical to kinetic.

I'd suggest to ask [here](https://github.com/RobotWebTools/rosbridge_suite) whether anything has changed between kinetic and melodic for the user.

Maybe other users who are already working with ROS melodic can provide better support?

hojjatabdollahi, 2019-02-22:
@MartinBischoff , thanks for your reply. We narrowed down the problem to the RosBridge installation. We were able to use RosSharp on a different ubuntu 18.04 and Melodic setup. However, uninstalling and purging ROS and RosBridge didn't help. We reinstalled Ubuntu and now everything is working. Thanks for your help.

MartinBischoff, 2019-02-26:
Happy to hear that. We are also upgrading to meldoic these days.
Closing.

closed 2019-02-26 by MartinBischoff