siemens/ros-sharp #166: Failed to connect to RosBridge

iulius2000, 2019-02-26:
* [x ] I am at the right place and my issue is directly related to ROS#. General technical questions I would post e.g. at [ROS Answers](https://answers.ros.org/) or [Stack Overflow](https://stackoverflow.com). For library-specific questions I would look for help in the corresponding library forums.
* [x ] I have thoroughly read [the Contributing Guideline](Contributing.md) and writing this issue is the right thing to do in my case.

---
## I have a question! ##

* [x ] I searched the [Wiki](https://github.com/siemens/ros-sharp/wiki), [open](https://github.com/siemens/ros-sharp/issues) and [closed](https://github.com/siemens/ros-sharp/issues?q=is%3Aissue+is%3Aclosed) issues for an answer. I tried my best to find the answer by myself without success. I believe that the discussion we will have in this issue, and the solutions we might find, will help me, and likely other community members who have a similar problem.

**Here is my question:**
I setup and run the ROS server on Ubuntu. When I hit play on Unity, I receive this error: Failed to connect to RosBridge at... 

---
## I found a bug! ##
* [x ] I am using the latest ROS# version available here on the master branch.
* [x ] I am adding all required information, code and data files, screenshots and log files so that you can reproduce the problem.

* My OS is:  Windows 10 
* My Unity Version is: 2018 3.6f1
* My ROS Distribution is: ROS Kinetic
* My Build Target Platform is: Windows

**Here is my bug description:**
 I setup and run the ROS server on Ubuntu. When I hit play on Unity, I receive this error: Failed to connect to RosBridge at... 

Perform the following steps reproduce the bug:
VMware Ubuntu: $ roslaunch unity_simulation_scene unity_simulation_scene.launch 
Unity: Play

Observed results:
Unity can't connect to RosBridge 

Expected results:

[image]
[image]

MartinBischoff, 2019-02-26:
Hi @iulius2000 ,
* In the RosConnector component, please switch to `websocket-sharp` protocol.
* Please also check whether the IP you entered is that of your VM and whether proxy, filewall or VM setting block the communication.

iulius2000, 2019-02-26:
@MartinBischoff 
I switched to websocket-sharp.
The IP is correct, but I am behind a company proxy. For both, Windows and Ubuntu  I used Cntlm to be able to access GitHub/internet behind the proxy. The default listening port for Cntlm is 3128, but I see that ROS is using 9090.

MartinBischoff, 2019-02-26:
Ì'd suggest to search for `Fatal error in GC: unexpected mark stack overflow`. I presume that some incompatibility between Unity and your computer system causes these errors.

iulius2000, 2019-02-27:
@MartinBischoff  I couldn't find the solution yet. 
I forgot to add something and maybe this helps: I use Unity and Ubuntu on the same computer.

iulius2000, 2019-02-27:
The problem was coming from VMware Network Adapter settings. I changed it from NAT to Custom (and I selected the NAT network).

closed 2019-02-27 by iulius2000

MartinBischoff, 2019-02-27:
thanks for this info @iulius2000 .
Is there anything we can improve in the [wiki page](https://github.com/siemens/ros-sharp/wiki/User_Inst_UbuntuOnOracleVM)?

iulius2000, 2019-02-27:
I used VMWare, not OracleVM.

Christyan593, 2020-05-12:
It works !!!!

The problem was coming from VMware Network Adapter settings. I changed it from NAT to Custom (and I selected the NAT network).