siemens/ros-sharp #168: Turtlebot2 cannot move using arrow keys

Xunzhaocunzi, 2019-02-26:
Hello,
I am working on "2.3 Gazebo Simulation Setup". The connection is set up successfully between ROS and Unity. ROS terminal can read the initial position of the turtlebot2 in Unity. 
However, I cannot move the turtlebot2 in Unity using arrow keys, like you did in the tutorial video. Is there a script file that I should attach it to the turtlebot2 in Unity to move it? Thank you.

MartinBischoff, 2019-02-27:
Sorry I can not reproduce this error. Please proceed as in the tutorial videos.

closed 2019-02-27 by MartinBischoff

luchspeter, 2020-03-22:
Hi
It seems that I have the same issue. 

This here is the current node map from rqt_graph 
[image]

I did check the topics with `rostipic echo /joy` which seems to work. The output of the python script "joy_to_twist" topic `/cmd_vel_mux/input/navi` seems to be published as well. But then the `mobile_base` node is missing and nothing feeds back into gazebo, which is why I guess there is not movement after pressing the buttons.

ThomasCarstens, 2020-03-30:
Hi @Xunzhaocunzi I am having the same issue, did you fix it?

luchspeter, 2020-03-30:
Hi @ThomasCarstens 
So far no luck no. But I'll have a few meetings with some "More-ROS-guys" than I am and maybe they will find the mistake.

We will also create some Docker Images for this ROS# / TurtleBot example that would not run on a Ubuntu VM. Maybe thats the problem. 

Are you experiencing the exact same issue? So do you get the same rqt_graph?

Regards
Eric

ThomasCarstens, 2020-03-30:
Here's my rqt. The most striking similarity I see is that gazebo only publishes the /camera/rgb (not the /joint_states nor /odom), and mobile_base_nodelet_manager is not publishing a /mobile_base to gazebo. 
[image]
I am using Ubuntu 16.04 on VMware and Unity 2019.3.0f5 on Windows.

MartinBischoff, 2020-03-31:
Hi @luchspeter and @ThomasCarstens ,
now that you have found it that this problem is not related to ROS#, I'd like to close this issue here.
For this problem you will find better help faster on e.g. [ROS Answers](https://answers.ros.org/questions/).
Please feel free to link this issue if you don't want to tell the story twice.
Thank you,
Martin