siemens/ros-sharp #169: Custom messages and features proposals

c-rizz, 2019-02-26:
Hi,
I've been using ROS# in Unity for a while now and I have a bunch of questions, I hope this is the appropriate place and way to ask.

The first one is about implementing custom messages. The wiki suggests to implement them inside the dll, but this doesn't seem to be necessary.  In my application I defined the messages keeping them separated from the ROS# code, and it seems to work fine. Is there a specific reason for not doing this?

Another point is that I needed to be able to setup the rosConnector at run time, changing the url based on user input. The current implementation doesn't seem to be designed for this, the only feature I was really missing was a way to wait for the connection to be established, so I had to add a method for it. If you are interest I could make a pull request for it.

There are also other things I could propose as pull requests, if you are interested:
 - Generic publisher and subscriber classes, which are not MonoBehaviours and don't need to be in the same gameObject as the rosConnector. They can just be instantiated anywhere in the code using a reference to the rosConnector and the message type, mimicking the ROS publishers and subscribers. This is particularly convenient when the rosConnector is instantiated dynamically.
- A simple NTP client which uses ROS messaging to get a time roughly synchronized with a different ROS node. Together with this, a message header which uses this time when updated. (This also has a ROS cpp side so it would be difficult to actually integrate in the library)
- A stub tf listener, it's not a fully featured tf listener, but it's a start.

Thanks

MartinBischoff, 2019-02-27:
Hi @c-rizz ,

happy to read that you got so involved with ROS#!
For better traceability, please write your questions in individual issues in the future.

__Custom Messages:__
You can put all sources of the RosBridgeClient and Urdf libraries in the Unity Assets folder and delete `RosbridgeClient.dll` and `Urdf.dll` from the Plugins folder instead. It will work. Sometimes I did that for debugging.

The reasons why we put the code into separate dlls is:
* The libraries can be used outside of Unity, e.g. in other game engines or applications.
* There are already quite a lot of scripts in the asset folder. The dlls help us cleaning things up a bit.

Feel free to put your custom messages and the libraries' sources wherever you want though.

__RosConnector at runtime__

We didn't need to set up RosConnector at runtime in any project yet. I set up the connection on `Start()`to keep things simple. If think your solution does not complicate the code and you are sure that there are no errors caused e.g. by publishers, subscribers, service providers or consumers that might require an established connection, we will take a look.

__Concerning the other extensions:__

Generally I am a big fan of keeping things simple in ROS#. The issue board is full of beginner questions and connection problems caused by the user or outside of Unity/ROS#. Complicating things, adding features that most users don't need and unfinished work will further increase our adminstration effort.
Understanding the simple version and the examples, people can easily add application-specific features by themselfes.

Nevertheless I very much appreaciate your extensions to ROS# and also your willingness to provide your work to the community. I set up a new issue where we can discuss about how to exchange / inform / advertise features that are not in the ros-sharp master.

c-rizz, 2019-03-25:
Hi,
Sorry for the super late reply, I've had some quite busy weeks.

For what concerns the RosConnector it's a really small thing, I actually just added a method that reads if the connection is established, you can see it here at line 87 https://github.com/OpenPTrack/optar/blob/noMarker/Assets/RosSharp/Scripts/RosBridgeClient/RosCommuncation/RosConnector.cs

About the generic publisher and subscribers, if you're interested you can see what I'm talking about here:
- https://github.com/OpenPTrack/optar/blob/noMarker/Assets/OpenPTrackCommons/scripts/GenericPublisher.cs
- https://github.com/OpenPTrack/optar/blob/noMarker/Assets/OpenPTrackCommons/scripts/GenericSubscriber.cs
- https://github.com/OpenPTrack/optar/blob/noMarker/Assets/OpenPTrackCommons/scripts/SafeGenericSubscriber.cs
The GenericSubscriber is not thread safe for interacting with unity objects, SafeGenericSubscriber is meant to tackle that.
Here you can see an example of their usage: https://github.com/OpenPTrack/optar/blob/noMarker/Assets/OpenPTrackCommons/scripts/NtpClient.cs

MartinBischoff, 2019-03-26:
Hi @c-rizz . None of your links work here.  Is https://github.com/OpenPTrack/optar/ private ?

berkayalpcakal, 2020-03-26:
Closed due to inactivity.

closed 2020-03-26 by berkayalpcakal