siemens/ros-sharp #174: Gazebo receives no data in Gazebo simulation scene

stevensu1838, 2019-03-05:
Hi all,

I've been following video ROS#: Gazebo Simulation Scene (for ROS# Version 1.2c) to do the same thing in Unity 2017.4. But I got the following rqt graph and the turtlebot doesn't move. Have any of you met this issue before? Thanks a lot

[image]

berkayalpcakal, 2019-03-05:
Hello @stevensu1838 ,

As far as It can be seen from rqt_graph, there should be no problem in terms of the communication framework. 

Have you tried monitoring the topics /joy and /joy_to_twist with command "rostopic echo /topic_name"?
You can also debug /joy messages on Unity side to see whether there exists a proper message flow. If you can observe that there exists a proper message flow in those topics, I suggest that you search for your issue on [https://answers.ros.org/questions/](url)

stevensu1838, 2019-03-08:
Hi All,

To any of you who experience the same error, the solution is not to set the following variables as instructed in the wiki [https://github.com/siemens/ros-sharp/wiki/User_Inst_TurtleBot2](url)

Set the required environment variables:
$ export TURTLEBOT_BASE=roomba
$ export TURTLEBOT_STACKS=circles
$ export export TURTLEBOT_3D_SENSOR=kinect

Then you don't see this communication error again. I don't know the exact reason but I solves my problems.  **My settings are  ROS# Version 1.2c, Unity 2017.4 and ROS indigo.**

closed 2019-03-08 by stevensu1838