siemens/ros-sharp #175: Adding a depth camera image publisher to ros-sharp

curranw, 2019-03-05:
I'm attempting to add a depth camera image publisher to ros-sharp, but I'm having a bit of an issue.  

I have a shader and material that turns the camera into a depth camera and renders the depth image. Here is an example for rendering the depth image with Unity.

```
public class DepthCamera : MonoBehaviour
{

    public Material mat;

    void Start()
    {
        GetComponent<Camera>().depthTextureMode = DepthTextureMode.Depth;
    }

    void OnRenderImage(RenderTexture source, RenderTexture destination)
    {
        Graphics.Blit(source, destination, mat);
    }
}
```
This works well, now I need to publish the image using ros-sharp. That is where I am having issues. I attempted to do this by updating the ImagePublisher class. Specifically I only updated the InitializeGameObject to apply my material to the 'source' RenderTexture (mat is the material and is an argument for the script). 

```
private void InitializeGameObject()
{
    texture2D = new Texture2D(resolutionWidth, resolutionHeight, TextureFormat.RGB24, false);
    rect = new Rect(0, 0, resolutionWidth, resolutionHeight);
    RenderTexture source = new RenderTexture(resolutionWidth, resolutionHeight, 24);
    Graphics.Blit(source, mat);
    ImageCamera.targetTexture = source;
}
```
This still simply publishes the RGB images to ROS. There must be something fundamental I am not understanding here.

MartinBischoff, 2019-03-06:
I guess the reason is that [Monobehaviour.OnRenderImage()](https://docs.unity3d.com/ScriptReference/MonoBehaviour.OnRenderImage.html) is called only after [Camera.OnPostRender()](https://docs.unity3d.com/ScriptReference/Camera.OnPostRender.html).

Have you tried modifying  ` ImagePublisher.UpdateMessage()`? Esp. I'd suggest to try calling `Graphics.Blit()` instead of the call at `DepthCamera.OnRenderImage()`.

Alternatively, you can call `UpdateMessage()` directly after `Graphics.Blit()` in your `OnRenderImage()`method.

I hope this helps! Please keep us informed on the progress you make.
Which shader and material are you using for your depth camera?

curranw, 2019-03-08:
Thanks for the message Martin. Yes I completely agree with the `OnRenderImage` being called after `OnPostRender`. That's why it's easy to visualize but difficult to send. 

A possible solution seems to be to chain these classes together (Unity camera -> DepthCamera -> ImagePublisher). As I wrote it, I think `DepthCamera` and `ImagePublisher` are both grabbing raw data from the unity camera. Not ideal.

`UpdateMessage()` is also called after `OnPostRender()`, [correct](https://github.com/siemens/ros-sharp/blob/master/Unity3D/Assets/RosSharp/Scripts/RosBridgeClient/RosCommuncation/ImagePublisher.cs#L41)? 

I tried a different technique, where I changed the active render texture (`RenderTexture.active = destination`) so that `readPixels()` uses my material. That also didn't work ([source](https://docs.unity3d.com/ScriptReference/Texture2D.ReadPixels.html)). 
 
Can you explain what you mean by: 

> Esp. I'd suggest to try calling Graphics.Blit() instead of the call at DepthCamera.OnRenderImage().

My Material is simply the default material with the following shader:

```
Shader "Custom/ZBuffer"
{
    SubShader
    {
        Tags { "RenderType"="Opaque" }
       
        Pass
        {
            CGPROGRAM
            #pragma vertex vert
            #pragma fragment frag
            #include "UnityCG.cginc"
 
            struct v2f
            {
                float4 pos : SV_POSITION;
                float4 screenuv : TEXCOORD1;
            };
           
            v2f vert (appdata_base v)
            {
                v2f o;
                o.pos = mul(UNITY_MATRIX_MVP, v.vertex);
                o.screenuv = ComputeScreenPos(o.pos);
                return o;
            }
           
            sampler2D _CameraDepthTexture;
 
            fixed4 frag (v2f i) : SV_Target
            {
                float2 uv = i.screenuv.xy / i.screenuv.w;
                float depth = 1 - Linear01Depth(SAMPLE_DEPTH_TEXTURE(_CameraDepthTexture, uv));
                return fixed4(depth, depth, depth, 1);
            }
            ENDCG
        }
    }
}
```

curranw, 2019-03-08:
Also tried the following for the DepthCamera to try to change the output feed. Didn't work.

```
public class DepthCamera : MonoBehaviour
{

    public Material mat;

    void Start()
    {
        GetComponent<Camera>().depthTextureMode = DepthTextureMode.Depth;
        Graphics.Blit(GetComponent<Camera>().activeTexture, GetComponent<Camera>().activeTexture, mat);
    }
}
```

DrWillway, 2019-05-03:
I had the same issue. Try publishing from LateUpdate. It should work

androclassic, 2019-09-03:
Hi, did you have any success with this ?

BhavyanshM, 2019-12-20:
> I had the same issue. Try publishing from LateUpdate. It should work

@DrWillway Could you please share the code on how you managed to simulate the depth camera and publish the depth images over ROS bridge? It would be really helpful for projects that requiring simulation of RGB-D cameras in Unity. Thank you so much.

berkayalpcakal, 2020-03-26:
Closed due to inactivity.

closed 2020-03-26 by berkayalpcakal