siemens/ros-sharp #176: Incorrect prismatic joint states

akashcastelino, 2019-03-07:
Hi,

In the URDF that I've imported from ROS to Unity, the joints consist of revolute and prismatic joints.

While all the revolute joint states are reproduced correctly when subscribed to the robot's joint states topic, the prismatic joints in the fingers are moving weirdly.

Here is a [link](https://drive.google.com/file/d/1N9QAvoCjonv4m2fsWhjEnsK78c3SAzGJ/view?usp=sharing) to the URDF used. It is the autogenerated URDF from the Moveit setup assistant (when using the panda_arm_hand.urdf.xacro from [the franka ros repository](https://github.com/frankaemika/franka_ros/tree/kinetic-devel/franka_description/robots) with inertial tags plus I added the transmission tags manually. 

Also, here is an [image](https://drive.google.com/file/d/11A8gqhVw5bwIcVgSBlAOjKjBHF5oTNhh/view?usp=sharing) showing the weird gripper prismatic joints after they are moved 

Could this be an issue with prismatic joints in ROS Sharp or otherwise I am not sure if this is an issue with the autogenerated URDF?

MartinBischoff, 2019-03-08:
Thank you for this execellent bug description @akashcastelino . It was easy to reproduce the error here.
Commit #178 should fix the issue.
Instead of local coordinates the prismatic joint translation had been applied in world coordinates before.

Please close this issue when done.

MartinBischoff, 2019-03-08:
@ReitDan fyi

akashcastelino, 2019-03-08:
Thanks for the bug fix !

closed 2019-03-08 by akashcastelino