siemens/ros-sharp #179: Gazebo simulation scene: The robot in Unity moves in opposite directions with the robot in Gazebo

stevensu1838, 2019-03-11:
Hi all
My settings are ROS# Version 1.2c, Unity 2017.4 and ROS indigo and I followed this video to run Turtlebot2 successfully. when I tried to do the same thing to **Tutlebot3**, everything goes well. Expect the robot in Unity **moves in opposite directions** with the robot in Gazebo. Like when the robot in Unity goes forward, the one in Gazebo moves backward. When the Unity robot turns left the gazebo one turns right. Any of you know how I can recalibrate it? Thanks a million.

anko0009, 2019-03-11:
Hi @stevensu1838 

I experienced the same issue when working with the TurtleBot2 in Unity. When I applied positive velocities, the robot moved in the opposite direction of Gazebo and a real TurtleBot2. 

I suppose that this issue somehow is related to the left-handed coordinate system in Unity. Even though the axis are changed accordingly when importing the URDF, the angular positive direction around the y-axis in the left-handed cs is opposite of the angular positive direction around the z-axis in the right-handed cs (does this make any sense?).

My workaround for also publishing the "right" values with the JointStateReader was:

1. Change the axis for the Hinge Joint in both actuated wheels to 0 -1 0 (instead of 0 1 0)
2. Add a minus in RosBridgeClient/MessageHandling/JointStateReader.cs to get the "negative" velocity. (velocity = -urdfJoint.GetVelocity();)

For my case this worked. Hope it helps.

stevensu1838, 2019-03-11:
Hi,
I first figured out it was a problem on **odom** instead of joint state. (I learnt this by not using join state message). Then I added negative in **PoseReceive.cs** in front of message.pose.pose.position.x and message.pose.pose.orientation.w as shown below. Thus, the problem solved. Hope it helps others. B ut who do you know why I experience this error? Thx a lot. Thank you for your help@anko0009

```
        private Vector3 GetPosition(NavigationOdometry message)
        {
            return new Vector3(
                -message.pose.pose.position.x,
                message.pose.pose.position.y,
                message.pose.pose.position.z);
        }

        
        private Quaternion GetRotation(NavigationOdometry message)
        {
            return new Quaternion(
                message.pose.pose.orientation.x,
                message.pose.pose.orientation.y,
                message.pose.pose.orientation.z,
                -message.pose.pose.orientation.w);
        }
    }
}
```

PymZoR, 2019-04-10:
Same problem here.
The above solution seems a bit "hacky" without knowing why in the first place those rotations are inverted from those shown in the video, given the same setup.

PymZoR, 2019-04-12:
Maybe I should open another issue to gather some attention (I don't know if anyone other than @stevensu1838 is getting notifications on this).

A simple way to reproduce this bug is to clone the repository, open the Gazebo simulation scene, and edit the two wheel joints in unity. 
- Tick `use motor`
- Tick `free spin`
- Enter a positive value for `target velocity`

Enter play mode : the turtle bot is moving backward (negative on the Unity z axis).

I've checked the conversion methods for Points / Vectors / Quaternions between ROS and Unity, which are all good. As mentioned in the first post, it might be related to the URDF import and how the joint axes are positioned. I'll try to dig further when I have some time.

MartinBischoff, 2019-05-20:
Dear @PymZoR, @anko0009  and @stevensu1838 ,

we delved into the problems you described above.
The root cause of the errors you are describing is that the ROS coordinate system is right-handed whereas the Unity coordinate system is left-handed.

This entails that the positive rotation direction in Unity is opposite to the positive rotation direction in ROS. In all problems you wrote about above, you are communicating angular rotation messages between ROS and Unity: angular positions, angular velocities, angular accelerations, torques, ...
You have to multiply -1 whenever you transfer those values between Unity and ROS.

We thought about pointing angular rotation axes into opposite directions, but this would be a hack. In case you change the joint into a prismatic joint, you would have to change the axis direction again. And in which axis direction should a cylindrical joint point to?

For `Vector3` and `Quaternion` we already have `Ros2Unity` and `Unity2Ros` methods in [TransformExtensions.cs](https://github.com/siemens/ros-sharp/blob/master/Unity3D/Assets/RosSharp/Scripts/Extensions/TransformExtensions.cs).
We did not add  corresponding transformation methods for angular float value, since since we found individually multiplying -1 is better than extening the `float`class with two methods `Ros2UnityAngular` and `Unity2RosAngular` which both effectively multiply the float value by -1.

I hope this clarifies your problems. I'm closing this issue. Please let me know if you know a better solution than this. Please reopen or answer here if anything is left unclear.

closed 2019-05-20 by MartinBischoff