siemens/ros-sharp #181: WebSocketNetProtocol reports it's alive after rosbridge_websocket is killed

missxa, 2019-03-26:
To reproduce:

```
// Unity app
void Start()
{
    string uri = "ws://192.168.0.113:9090";
    ws = new WebSocketNetProtocol(uri);
    rosSocket = new RosSocket(ws);
}

void Update() 
{
    UnityEngine.Debug.Log(rosSocket.protocol.IsAlive());
}
```

1. `roslaunch rosbridge_server rosbridge_websocket.launch`
2. Start Unity 
( Unity debugger shows `True`)
3. Kill rosbridge
( Unity debugger still shows `True`)

I tested `rosSocket` with `WebSocketSharpProtocol` and it works as expected. 
Is it a bug or implementation specifics? What is the proper way to check if connection is still alive for `WebSocketNetProtocol`?

MartinBischoff, 2019-03-27:
Hi @missxa !
* How did you "3. Kill rosbridge"?
* It looks like a bug in our [WebSocketNetProtocol](https://github.com/siemens/ros-sharp/blob/master/Libraries/RosBridgeClient/Protocols/WebSocketNetProtocol.cs) class. We wrote this client for ROS# as an alternative to  [websocket-sharp](https://github.com/sta/websocket-sharp) since websocket-sharp does not support .NET Core.

The `IsAlive()` Method:
```
      public bool IsAlive()
        {
            return clientWebSocket.State == WebSocketState.Open;
        }
```
obviously doesn't work as it should:
* Either because comparing with `WebSocketState.Open` is not correct. [Here](https://docs.microsoft.com/de-de/dotnet/api/system.net.websockets.websocketstate?view=netframework-4.7.2) is an overview on the available WebSocketStates.
* Or because `clientWebSocket`has not been informed about the state change in the way you shout down the Websocket. This is unlikely since you wrote that WebsocketSharpProtocol already works as expected; i.e. the procols are already sufficiently informed.

Please let me know if you find a better `IsAlive()` implementation that works for you.

MartinBischoff, 2020-04-21:
Hi @missxa ! Do you have any update on this issue?

missxa, 2020-04-21:
I believe I used `WebSocketSharpProtocol` in the end

closed 2020-04-21 by missxa