siemens/ros-sharp #183: ERROR: Import URDF from ROS

DengMark, 2019-04-01:
* [ ] I am at the right place and my issue is directly related to ROS#. General technical questions I would post e.g. at [ROS Answers](https://answers.ros.org/) or [Stack Overflow](https://stackoverflow.com). For library-specific questions I would look for help in the corresponding library forums.
* [ ] I have thoroughly read [the Contributing Guideline](Contributing.md) and writing this issue is the right thing to do in my case.

---
## I have a question! ##

* [ ] I searched the [Wiki](https://github.com/siemens/ros-sharp/wiki), [open](https://github.com/siemens/ros-sharp/issues) and [closed](https://github.com/siemens/ros-sharp/issues?q=is%3Aissue+is%3Aclosed) issues for an answer. I tried my best to find the answer by myself without success. I believe that the discussion we will have in this issue, and the solutions we might find, will help me, and likely other community members who have a similar problem.

**Here is my question:**

My OS is: Windows 7 64bit

My Unity Version is: 2018 3.11f1

My ROS Distribution is: ROS Kinetic

My Build Target Platform is: Windows

Unity is run in Windows while ROS run in another machine with ubuntu system. There is some errors when I follows the tutorial 2.1.

    $ roslaunch file_server publish_description_turtlebot2.launch
    ... logging to /home/zhenglong/.ros/log/e4123a80-5487-11e9-a987-8ca982cd582f/roslaunch-MobileStation-32636.log
    Checking log directory for disk usage. This may take awhile.
    Press Ctrl-C to interrupt
    Done checking log file disk usage. Usage is <1GB.

    started roslaunch server http://MobileStation:45709/

    SUMMARY
    ========

    PARAMETERS
    * /robot/name: Turtlebot2
    * /robot_description: <?xml version="1....
    * /rosapi/params_glob: [*]
    * /rosapi/services_glob: [*]
    * /rosapi/topics_glob: [*]
    * /rosbridge_websocket/address: 
    * /rosbridge_websocket/authenticate: False
    * /rosbridge_websocket/bson_only_mode: False
    * /rosbridge_websocket/delay_between_messages: 0
    * /rosbridge_websocket/fragment_timeout: 600
    * /rosbridge_websocket/max_message_size: None
    * /rosbridge_websocket/params_glob: [*]
    * /rosbridge_websocket/port: 9090
    * /rosbridge_websocket/retry_startup_delay: 5
    * /rosbridge_websocket/services_glob: [*]
    * /rosbridge_websocket/topics_glob: [*]
    * /rosbridge_websocket/unregister_timeout: 10
    * /rosbridge_websocket/use_compression: False
    * /rosdistro: kinetic
    * /rosversion: 1.12.14

    NODES
    /
        file_server (file_server/file_server)
        rosapi (rosapi/rosapi_node)
        rosbridge_websocket (rosbridge_server/rosbridge_websocket)

    auto-starting new master
    process[master]: started with pid [32652]
    ROS_MASTER_URI=http://localhost:11311

    setting /run_id to e4123a80-5487-11e9-a987-8ca982cd582f
    process[rosout-1]: started with pid [32665]
    started core service [/rosout]
    process[rosbridge_websocket-2]: started with pid [32682]
    process[rosapi-3]: started with pid [32683]
    process[file_server-4]: started with pid [32684]
    [ INFO] [1554127824.695614255]: ROSbridge File Server initialized.
[... 268 lines cut ...]
        value = future.result()
    File "/usr/lib/python2.7/dist-packages/tornado/concurrent.py", line 215, in result
        raise_exc_info(self._exc_info)
    File "/usr/lib/python2.7/dist-packages/tornado/gen.py", line 956, in handle_yield
        self.future = convert_yielded(yielded)
    File "/usr/lib/python2.7/dist-packages/tornado/gen.py", line 1026, in convert_yielded
        raise BadYieldError("yielded unknown object %r" % (yielded,))
    BadYieldError: yielded unknown object None

    [ INFO] [1554127873.048645914]: get_file request: package://turtlebot_description/meshes/sensors/kinect.jpg

    [ERROR] [1554127873.051248]: Traceback (most recent call last):
    File "/opt/ros/kinetic/lib/python2.7/dist-packages/rosbridge_server/websocket_handler.py", line 168, in prewrite_message
        yield self.write_message(message, binary)
    File "/usr/lib/python2.7/dist-packages/tornado/gen.py", line 870, in run
        value = future.result()
    File "/usr/lib/python2.7/dist-packages/tornado/concurrent.py", line 215, in result
        raise_exc_info(self._exc_info)
    File "/usr/lib/python2.7/dist-packages/tornado/gen.py", line 956, in handle_yield
        self.future = convert_yielded(yielded)
    File "/usr/lib/python2.7/dist-packages/tornado/gen.py", line 1026, in convert_yielded
        raise BadYieldError("yielded unknown object %r" % (yielded,))
    BadYieldError: yielded unknown object None

    [INFO] [1554127873.348038]: Client disconnected. 0 clients total.

anko0009, 2019-04-04:
Hi @DengMark,

I ran into the same error when running my simulation. I think it's an issue in the rosbridge_suite and - at least for me - it seems to be not influencing the functionality. There is an issue describing this in the github repo of the rosbriduge_suite (https://github.com/RobotWebTools/rosbridge_suite/issues/394). Seems like it has been fixed and will be synced soon.

DengMark, 2019-04-08:
> Hi @DengMark,
> 
> I ran into the same error when running my simulation. I think it's an issue in the rosbridge_suite and - at least for me - it seems to be not influencing the functionality. There is an issue describing this in the github repo of the rosbriduge_suite ([RobotWebTools/rosbridge_suite#394](https://github.com/RobotWebTools/rosbridge_suite/issues/394)). Seems like it has been fixed and will be synced soon.

Thank you for your reply. The error has been solved after updating the rosbridge_suite (newly update to v-0.11.0).

closed 2019-04-08 by DengMark