siemens/ros-sharp #184: ROS# on Magic Leap

dmitterberger, 2019-04-03:
* My OS is: 
* My Unity Version is: 
* My ROS Distribution is: 
* My Build Target Platform is: 

---
## I have a question! ##

**I want to do this:**
Use ROS# on Magic Leap

We are trying to get to run the ROS# websocket on the Magic Leap. We can run the socket when connected to Unity, but when building the app onto the Magic Leap Platform, we are sucessful in connecting to the rosbridge via RosConnector, but we are not able to subscribe to a topic.

In Subscriber.cs it stops here:
 GetComponent<RosConnector>().RosSocket.Subscribe<T>(Topic, ReceiveMessage, (int)(TimeStep * 1000));

We cannot figure out if it is a problem with the protocol, or the built-in firewall of the Magic Leap. We made sure (via privilege requester) that all privileges for LAN are given.
Has anybody ever tried building onto a MagicLeap directly (not via simulator)?

dmitterberger, 2019-04-12:
After quite a bit of digging I realized that the problem was not the permissions nor the firewall but with the serialization with Newtonsoft Json.NET, which didnt work when building and using for IL2CPP (as it is an ahead of time compilation). After replacing the regular Json.NET with Newtonsoft Json.NET AOT (using this Unity package: https://assetstore.unity.com/packages/tools/input-management/json-net-for-unity-11347) and changing some lines in communication.cs, build rossharp.sln and update the rosbridgeclient.dll in my unity project the connection worked and I successfully talk to the Magic Leap using ROS#.

MartinBischoff, 2019-04-12:
Thanks @dmitterberger for reporting the solution to the problem you had with the Magic Leap.
Is there anything that you would suggest to include upstream in ROS#? e.g. the JSON.NET for Unity asset or your changes in communication.cs?

dmitterberger, 2019-04-12:
regarding the changes in communications.cs.. instead of using _public override string op { get { return "advertise"; } }_  simply using _this.op ="advertise"_ etc.. Regarding the Json.net solved follow up Jason issues with the specific Json.net for Unity. Regarding Json.net I planned to upload a forked project for Magic Leap users who would like to implement ROS#

MartinBischoff, 2019-04-12:
Thank you @dmitterberger for the the pull request #190 . How shall we proceed with the Json.NET.
We could include the Json.NET unity asset into the Unity3D project and into the RosSharp.unitypackage instead of [Newtonsoft.Json.dll](https://github.com/siemens/ros-sharp/blob/master/Unity3D/Assets/RosSharp/Plugins/Newtonsoft.Json.dll).
Suggestions?

gonzalocasas, 2019-04-15:
@dmitterberger @MartinBischoff Perhaps the right assembly could be conditionally included based on the selected configuration using something like `Condition=" '$(Configuration)|$(Platform)' == 'IL2CPP' "`? Someone would need to define configurations first, because iirc, they are currently default ones.

dwhit, 2019-04-16:
I had to replace the newtonsoft json with the magic leap compatible version for a project, and it was able to be a drop in replacement, no conditions needed.

shloksobti, 2019-05-21:
I'm having trouble going through this with the latest Ros sharp version. All I'm supposed to do is switch out the json.dll file? What about the bson files?

MartinBischoff, 2019-05-22:
@akirayou can you help here?

akirayou, 2019-05-22:
Sorry I don't know about magic leap device , may be @tarukosu  knows.

shloksobti, 2019-05-22:
Just to elaborate a little bit. Everything works perfectly fine in the previous version of ROS Sharp, assuming the environment settings as described earlier in the thread. However, switching to the latest commit, I am no longer able to subscribe/publish from the device to ROS topics. 

I could always use the earlier version - but wanted to be sure that there is no easy fix to this. Should probably also raise a separate issue concerning this is that's the case. Thank you so much for your help.

MartinBischoff, 2019-05-23:
I presume we are destroying UWP compatibility with [this pull request](https://github.com/siemens/ros-sharp/pull/198), in particular with the [Newtonsoft.Json.Bson.dll](https://github.com/siemens/ros-sharp/blob/master/Unity3D/Assets/RosSharp/Plugins/Newtonsoft.Json.Bson.dll).

We do not develop or test for UWP here. This is the reason why @dwhit set up a UWP version of ROS# [here](https://github.com/dwhit/ros-sharp).

@dwhit  had to replace the Newtonsoft.Json.dll with a UWP-compatible version as discussed above. I think the same we will have to do for the Newtonsoft.Json.Bson.dll.

dwhit, 2019-05-23:
Yes, that is correct, on my UWP branch I have a UWP compatible bson dll.

shloksobti, 2019-05-23:
Thanks for your reply! I just tried dropping in @dwhit RosSharp folder into my project replacing the earlier one. And although I am able to connect to the WebSocket, I am unable to subscribe to ROS topics. Am I missing something? Have you tested it with the MagicLeap yet?

dwhit, 2019-05-23:
Magic leap? The UWP branch is for the HoloLens. I haven’t tried that branch with the Magic Leap.

shloksobti, 2019-05-23:
Oh alright, I thought so! Thanks! It seems like there isn't a drop-in solution specific to the MagicLeap yet.

gonzalocasas, 2019-05-23:
@shloksobti well, @dmitterberger has it working on the magicleap, did you try her suggestion to replace the Json.net dll?

shloksobti, 2019-05-23:
@gonzalocasas Oh yeah, I got that working with the older version of ROSSharp. I was trying to update all the components in my project and hence trying to get something similar working with the latest commit.

shloksobti, 2019-05-23:
@dmitterberger have you had any chance to try out the latest commit of ROSSharp?

akirayou, 2019-05-23:
FYI:
Just putting link.xml may be enough .
In this tweet, he uses ROS# on Hololens  with the link.xml.
But I don't tried it yet .Anyone ,have you tried about IL2CPP bytecode stripping settings?

https://twitter.com/tarukosu/status/1131350438509367296
https://docs.unity3d.com/Manual/IL2CPP-BytecodeStripping.html

nafajardo, 2019-05-24:
Hello! Just to raise awareness, I am also trying to get my Magic Leap One device running with ROS Sharp. So far I can connect after building once and that is it. It can publish and subscribe during that magic window. 

I believe I just need to read in depth what you guys are doing, but if the latest commit has the fixes ill just download that.

shloksobti, 2019-05-24:
@nafajardo what version of ROS# are you using?
Also, I think the reason you're having this problem is due to Unity privileges. On the device, go to Settings->Privacy->Priviliges and grant LAN privilege to your app. Hopefully, this fixes it.

nafajardo, 2019-05-24:
ROS# Version as of [this commit](https://github.com/siemens/ros-sharp/tree/eb096c7bdf3a31f432013663f05626d33aa59b41). I haven't done that yet, will let you know how it goes.

shloksobti, 2019-05-24:
Oh, I tried using that but wasn't able to get any subscriptions. Did you change anything? Such as the JSon? or BSon? Would you mind elaborating more on your setup? I've had to revert back to v1.3 to be able to successfully setup subscriber and publishers.

nafajardo, 2019-05-24:
I just used the stock code, no changes. Currently just communicating over a router to a computer locally. In terms of privileges, I do not see it in the device under LAN. After reading some documentation it seems like my manifest needs to be edited to do this.

shloksobti, 2019-05-24:
Oh, you haven't yet checked on MagicLeap? Just Unity? I am able to get this commit working and communicating with Unity; however, once I build to the device, subscribers and publishers do not work.

shloksobti, 2019-05-24:
@nafajardo Yes, you'd have to add LAN in the manifest. 
    <uses-privilege ml:name="LocalAreaNetwork" />

nafajardo, 2019-05-24:
When I build and run my project to the device, it communicates with ROS. Once I close and open the application again it doesn't connect.

shloksobti, 2019-05-24:
Yeah, I'm confident that's a privilege issue. I'm not able to get it to subscribe via the device for some reason when using the latest version of ROS#

nafajardo, 2019-05-24:
Yep works over LAN now thank you! I am about to unit test my publishers and subscribers. FYI I am using kinetic on 16.04 if that matters. If my publishers and subs work I will come back and send some generic files your way to test.

shloksobti, 2019-05-24:
Awesome! I'm glad you got it working. Yeah, I'll be forever grateful ha!

nafajardo, 2019-05-24:
Tested: In short, the subscribers and publishers only successfully work when using the Zero Iteration software because of the connection VIA the Unity Workstation (Not the MLO device itself). We are able to connect to the router but not able to publish or subscribe to anything when running the app on the device itself. So now I am looking into the changes mentioned above :/

shloksobti, 2019-05-24:
Yeah, that's pretty much what we faced! I suggest going back to v1.3, and using the external JSON package mentioned above. Let me know if you need help.

nafajardo, 2019-05-24:
Thank you! I just reverted to 1.3, and downloaded that JSON package. Could you elaborate on what to do next?

shloksobti, 2019-05-24:
It'll be a good idea to take this conversation elsewhere. Shoot me an email and we can go from there.

nafajardo, 2019-05-24:
Sounds good. Just shot you an email.
To all the others here, thanks for bringing this issue up. It seems like there is a good amount of interest in AR/VR development with unity and ROS applications. Would it be a plausible idea to open a separate branch from master to cater to these applications? If not has someone forked a repo that is catered to AR/VR. Either way, Ill be keeping my eyes peeled on this thread.

nafajardo, 2019-05-24:
Or perhaps branches for specific technologies..
Master
|-> UWP
|-> Magic Leap
|-> Hololense

shloksobti, 2019-05-24:
@MartinBischoff What're your thoughts on this?

akirayou, 2019-05-24:
I investigated about this problem a little bit.
This problem caused by "System.Reflection.Emit" .
System.Reflection.Emit is key technology for fast parsing.But It not good for IL2CPP.
https://docs.unity3d.com/2018.3/Documentation/Manual/ScriptingRestrictions.html

It sounds good to fork for the library ( Fast but can NOT run on IL2CPP  / Slow but can run on IL2CPP )
For example JSON.net for unity can run on IL2CPP.
https://assetstore.unity.com/packages/tools/input-management/json-net-for-unity-11347

Almost done with just replace library (i.e. Visual studio project settings).
But be careful, JSON.net for unity have old style BSON support.
ROS# uses new style BSON support.So you have to rewrite few lines in RosSocket.cs

//old style
https://www.dustinhorne.com/post/2013/09/25/Json-NET-for-Unity-Developers
>     var serializer = new JsonSerializer();
>     var writer = new BsonWriter(stream);
>     serializer.Serialize(writer, p);    
> 

//new style
https://github.com/siemens/ros-sharp/blob/2bcb2666a5af3222da840fd4e87b0a99700fb068/Libraries/RosBridgeClient/RosSocket.cs#L185-L217
>   System.IO.MemoryStream ms = new System.IO.MemoryStream();  
>   Newtonsoft.Json.Bson.BsonDataWriter writer = new Newtonsoft.Json.Bson.BsonDataWriter(ms);
>   JsonSerializer serializer = new JsonSerializer();
>   serializer.Serialize(writer, obj);
> 

OK,Now I understood.But I Dot't have UWP device...good luck.

MartinBischoff, 2019-05-27:
> @MartinBischoff What're your thoughts on this?

We are currently not using UWP, Magic Leap or Hololens here and are not planning to.
Someone interested in any particular AR/VR technology / platform support please fork this repo and prepare a platform-specific ROS# version as @dwhit already did for UWP [here](https://github.com/dwhit/ros-sharp).
No forks. They will diverge and become a mess on the long run.

cerenoozturk, 2019-07-20:
I am sorry for duplicate question but I am new to Unity and  I have nearly the same problem. I built Unity application to my Iphone and I am trying to connect ROS to Unity application. They are connecting but my application cant publish any node. For this problem, what should I do? I import Json.Net and I am choosing IOS option but do i need to change any file inside the Json.Net?

akirayou, 2019-07-21:
Have you checked using .Net 4.x (Not .Net Starndard 2.0) in player setting?

cerenoozturk, 2019-07-21:
Now, I changed it to .Net 4.x. But now it is giving "Unloading broken assembly Assets/RosSharp/Plugins/RosBridgeClient.dll, this assembly can cause crashes in the runtime" error.

berkayalpcakal, 2020-03-26:
Closed due to inactivity. 

Hi @cerenoozturk, maybe you can open another issue using the issue template?

closed 2020-03-26 by berkayalpcakal

Bradyk27, 2020-06-08:
@dmitterberger Could you clarify the steps you took to communicate the Magic Leap with ROS#? I'm afraid I'm a bit lost regarding what pull requests were accepted & what not.

My understanding is that I need to import [this asset](https://assetstore.unity.com/packages/tools/input-management/json-net-for-unity-11347) and that the communication.cs file _has_ since been edited to allow for support. Are there any other steps you took to make this communication possible? Thanks.

aioniosfoititis, 2020-06-09:
@Bradyk27  I have not used magic leap ,but I will get one probably in the future.

Although In the Hololens version in order to fix the similar problem mentioned by you you should switch the  json files inside rossharp and change them with the AOT version that you improrted.

Did you tried that? let me know if it works and it is a similar fix.

Bradyk27, 2020-06-28:
@aioniosfoititis JSON files have been replaced, receiving this error in Unity: 

`Unloading broken assembly Assets/RosSharp/Plugins/RosBridgeClient.dll, this assembly can cause crashes in the runtime
`
Any way to go about rebuilding RosBridgeClient.dll?

Bradyk27, 2020-07-27:
Hi everyone, here's my final update outlining exactly what I did to get ROS, ROS#, and Magic Leap to talk together.

1. I started by pulling #190 as my ROS# base (using newer versions of ROS# causes compile problems with BSON files.) 
2. Import the Unity asset RosSharp folder into your project (make sure project is setup with proper Lumen tools, drivers, etc. See [Magic Leap documentation](https://developer.magicleap.com/en-us/learn/guides/get-started-developing-in-unity) for this 
3. In Edit->Project Settings->PlayerSettings, make sure NET compatibility is set to 2.0.
4. Download the [Unity Asset Newtonsoft JSON.net ](https://assetstore.unity.com/packages/tools/input-management/json-net-for-unity-11347). This is going to cause name-errors so I'd suggest importing it in a different project and then just pulling files from finder / file-explorer rather than importing it directly. Unity is dumb.
5. Under the AOT folder in the Newtonsoft JSON.net asset, copy the Newtonsoft.json.dll file. Replace the Newtonsoft.json.dll file in the RosSharp asset (Plugins folder) with it. Might need to refresh / reimport assets. 
6. Open the RosSharp.sln project in Visual Studio (found under libraries in the full Github project)
7. Clean & Rebuild (will need to install NET 4.6.0 if you don't already have it. Follow VS popup notifications.) We are doing this because communication.cs has been changed in this version as needed but RosBridgeClient.dll has not. So we are rebuilding with the updated communication.cs to get the updated RosBridgeClient.dll, which we will then import into unity.
8. The rebuild will output a *new* RosBridgeClient.dll in the bin folder (use the search to find it if you're not sure of where the bin folder is. Note the timestamp on the file, it should be whatever time the rebuild finished).
9. Replace RosBridgeClient.dll in the UNITY project (Plugins folder) with this newly created RosBridgeClient.dll from the build output.
10. Refresh, reimport assets.
11. On the RosConnector object in Unity, make sure to switch from SHARP to NET protocol. This drop down can be found in the inspector. Now we are using the NET protocol with updated Newtonsoft.json.dll and accompanying RosBridgeClient.dll.
12. Refresh / reimport assets. Build, run on Leap. Should work fine. If having permission errors, make sure the Magic Leap has granted LAN privileges to your application (see a little above in this issue forum).

Took me a month to figure all of this out but hey better late than never.

MartinBischoff, 2020-07-28:
Hi @Bradyk27 ,
thank you for this thorough description. Aiming for a preferably high multi-platform support, we will try to adapt future ROS# versions to reduce the number of steps and simplify this procedure.

Bradyk27, 2020-07-30:
@MartinBischoff No problem, big thanks to my uni for paying me while I figured it out :) 

Insofar as simplifying procedure, the main problem I ran into is RosBridgeClient.dll being unloaded upon replacement of the JSON file with the appropriate one, mainly due to errors with the BSON objects (not JSON). Hence why using an old version of ROS# (that did not have BSON support) worked. 

Not sure how all of this communicates together but apparently they are somehow interlocked. I think the error comes from the JSON being UWP-compatible and the BSON not. I believe that replacing *both* the JSON and BSON file with UWP-compatible files will work (although it might take recompilation of RosBridgeClient.dll), although I'm not sure where a compatible BSON can be found. 

I'm glad to help with any questions anyone has following this process, direct them here or send me an email.