siemens/ros-sharp #185: AckermannDriveStamped Message

rukie, 2019-04-03:
We are struggling to get the [Ackermann Drive Stamped message ](http://docs.ros.org/melodic/api/ackermann_msgs/html/msg/AckermannDriveStamped.html)  to play with Unity/ROS#. We are on Unity 2018.03f (Windows) and ROS Kinetic (Ubuntu 16.04 VM).

We have [Ackermann Drive messages] (http://docs.ros.org/melodic/api/ackermann_msgs/html/msg/AckermannDrive.html) functioning. (See below)

Is there an issue with the way ROS Sharp deserializes the messages? I would appreciate feedback in the approach below. Thank you!

Ackermann Drive Publisher
```C#
using UnityEngine;
using System.Collections;
using RosSharp.RosBridgeClient.Messages.Ackermann;

namespace RosSharp.RosBridgeClient
{
    public class AckermannDrivePublisher : Publisher<Messages.Ackermann.AckermannDrive>
    {
        private AckermannDrive ackermannDrive;

        private float steeringAngle;
        private float steeringAngleVelocity;
        private float speed;
        private float acceleration;
        private float jerk;

        protected override void Start()
        {
            base.Start();
            InitializeAckermannDrive();
        }

        private void InitializeAckermannDrive()
        {
            ackermannDrive = new AckermannDrive();
        }

        public void UpdateMessage(Messages.Ackermann.AckermannDrive ackermannDrive)
        {
            steeringAngle = ackermannDrive.steering_angle;
            steeringAngleVelocity = ackermannDrive.steering_angle_velocity;
            speed = ackermannDrive.speed;
            acceleration = ackermannDrive.speed;
            jerk = ackermannDrive.jerk;

            Publish(ackermannDrive);
        }
    }
}

```
AckermannDriveStampedPublisher
```C#
using UnityEngine;
using System.Collections;
using RosSharp.RosBridgeClient.Messages.Ackermann;

namespace RosSharp.RosBridgeClient
{
    public class AckermannDriveStampedPublisher : Publisher<Messages.Ackermann.AckermannDriveStamped>
    {
        private AckermannDriveStamped ackermannDriveStamped;

        private Messages.Standard.Header header;
        private AckermannDrive ackermannDrive;
        
        protected override void Start()
        {
            base.Start();
            InitializeAckermannDriveStamped();
        }

        private void InitializeAckermannDriveStamped()
        {
            ackermannDriveStamped.header = new Messages.Standard.Header();
            ackermannDriveStamped.ackermannDrive = new AckermannDrive();
        }

        public void UpdateMessage(Messages.Ackermann.AckermannDriveStamped ackermannDriveStampedTemp)
        {
            ackermannDriveStamped.header = ackermannDriveStampedTemp.header;
            ackermannDriveStamped.ackermannDrive = ackermannDriveStampedTemp.ackermannDrive;

            Publish(ackermannDriveStamped);
        }
    }
}
```

anko0009, 2019-04-03:
Hi @rukie,

I didn't really get what your exact issue was in getting the publisher to "play" - what errors do you have?
By having a brief look over the code you posted, I noticed that you don't initialize the stamped version of the ackermanndrive, just its "child". Compare your code with the PoseStampedPublisher in RosSharp, here the posestamped is initialized instead of the pose.
I also noticed that you don't call your UpdateMessage void, but I'm not an expert in C# and maybe you can do that from somewhere else. If it is not on purpose, you might want to check that as well.

Hope this helps you a bit.

rukie, 2019-04-03:
Oops.
private AckermannDrive ackermannDrive;
When looking for data in ROS, the data from AckermannDriveStamped.AckermannDrive is empty. Its null. 
I can send messages with just ackermann drive, but I when I send ackermanndrivestamped I only get the header information. I don't get the steer angle/velocity/etc.

osholsonel20, 2019-04-03:
Working with @rukie on this, I figured out the issue. The Ackermann Drive Stamped that I had implemented in ROS Sharp did not have the correct variable name. 

The message is:
Header header
AckermannDrive drive

I had implemented:

        public AckermannDriveStamped()
        {
            header = new Standard.Header();
            ackermannDrive = new AckermannDrive();
        }

It should have been drive not ackermannDrive. Silly mistake but oh man that had me questioning my sanity for hours.

MartinBischoff, 2019-04-04:
@osholsonel20 thanks for reporting. Happy to read that you made progress!
If your issue is completely resolved now, please close it.

closed 2019-04-04 by rukie