siemens/ros-sharp #191: ROS# with UDP

bomarali, 2019-04-12:
(confirm) I am at the right place and my issue is directly related to ROS#. General technical questions I would post e.g. at [ROS Answers](https://answers.ros.org/) or [Stack Overflow](https://stackoverflow.com). For library-specific questions I would look for help in the corresponding library forums.
(confirm) I have thoroughly read [the Contributing Guideline](Contributing.md) and writing this issue is the right thing to do in my case.

---
## I have a question! ##

* [ ] I searched the [Wiki](https://github.com/siemens/ros-sharp/wiki), [open](https://github.com/siemens/ros-sharp/issues) and [closed](https://github.com/siemens/ros-sharp/issues?q=is%3Aissue+is%3Aclosed) issues for an answer. I tried my best to find the answer by myself without success. I believe that the discussion we will have in this issue, and the solutions we might find, will help me, and likely other community members who have a similar problem.

**Here is my question:**
<Hi everyone!
I am setting up a communication between Unity and ROS using two different machines over static IP (at least I believe I am trying to do that). So I set my IPv4 settings on the Ubuntu to a static address, set my Windows machine to other static ip, connected them with LAN and have attempted to run the tutorial 2.1 (sending urdf from ros to unity) and it failed in exactly same fashion as https://github.com/siemens/ros-sharp/issues/71 or https://github.com/siemens/ros-sharp/issues/149>, i.e. it procs connect/disconnect on both sides and nothing more. 
I am aware that the tutorials were meant for VM but it does not suit my project. I have tried to run the code that was suggested in 71 but I guess the api changed since so I am trying to rewrite it now. Although judging from other issues the problem seems to be rooted in communications. I could not find other attempts to run the ros# with static ip, hence the new issue. Ideas?

---
* My OS is: <Windows 10, Ubuntu 16.04 on 4.14.12 rt kernel>
* My Unity Version is: <2018.3>
* My ROS Distribution is: <ROS Kinetic>
* My Build Target Platform is: <TBD>

**I want to do this:**
<Connect two machines>

bomarali, 2019-04-12:
I have disabled the firewall and the anti-virus, it did not help. I have echo'ed the connected clients on the ros side and saw that the windows machine was connected for short period (~100ms) and disconnected. In that little period nothing was published. (I had a pose publisher running from windows)

bomarali, 2019-05-01:
Is there a difference between enp0s31f6 and enp0s8?

MartinBischoff, 2019-06-07:
Hi @bomarali ! I am sorry noone found time to support you with your issue for such a long period of time.
Have you already proceeded in solving your problem yourself in the meantime?

> Is there a difference between enp0s31f6 and enp0s8?

Both seem to be [Predictable Network Interface Names](https://www.freedesktop.org/wiki/Software/systemd/PredictableNetworkInterfaceNames/) and just specify which network interface you are using on your machine.

Based on your problem descriptions above it seems to me like a pure LAN connection problem. RosbridgeClient connects to rosbridge_suite via a common WebSocket connection. These work on both LAN and WLAN connections with dynamic or static IPs, between virtual and real machines. Make sure that both machines are in the same network and can communicate.

> Recommended feature
I believe there is a need for a basic tutorial on how to setup simple subscriber publisher class in unity with ros# since the code suggested in 71 does not work and points to a bit of wiki that does not exist

We removed the wiki page since we included the code [here in the repo](https://github.com/siemens/ros-sharp/blob/master/Libraries/RosBridgeClientTest/RosSocketConsoleExample.cs).

You find more console and test classes  that run without Unity in [RosBridgeClientTest](https://github.com/siemens/ros-sharp/tree/master/Libraries/RosBridgeClientTest) and [UrdfTest](https://github.com/siemens/ros-sharp/tree/master/Libraries/UrdfTest).

bomarali, 2019-06-13:
Thank you for the reply. I did not manage to make it work, so I am currently using own communication setup. That said, I really want to make ros-sharp work. 
It still fails to connect to the server, even in non-unity version. Also the VS solution jumps right into the urdf export to ros in the rosbridgeclienttest. I am new to c# VS framework. Could you tell me how to only run the RosSocketConsoleExample?

bomarali, 2019-06-14:
(Resolved) So it was hardware issue all along. 
I have played with roslibjs around trying to find where connection is dropped. After a while a saw some inconsistencies on my server laptop. Switched it with my client VR PC. Everything works as intended. I am both relieved and incredible frustrated. I should have tested for hardware issues much earlier.

closed 2019-06-14 by bomarali