siemens/ros-sharp #193: ROS# UWP Build

alaksana96, 2019-04-16:
* [x] I am at the right place and my issue is directly related to ROS#. General technical questions I would post e.g. at [ROS Answers](https://answers.ros.org/) or [Stack Overflow](https://stackoverflow.com). For library-specific questions I would look for help in the corresponding library forums.
* [x] I have thoroughly read [the Contributing Guideline](Contributing.md) and writing this issue is the right thing to do in my case.

---
## I have a question! ##

* [x] I searched the [Wiki](https://github.com/siemens/ros-sharp/wiki), [open](https://github.com/siemens/ros-sharp/issues) and [closed](https://github.com/siemens/ros-sharp/issues?q=is%3Aissue+is%3Aclosed) issues for an answer. I tried my best to find the answer by myself without success. I believe that the discussion we will have in this issue, and the solutions we might find, will help me, and likely other community members who have a similar problem.

**Here is my question:**

Hi All,

I am currently trying to build @dwhit fork of the ROS# project for the Hololens. Unfortunately, I am having some trouble building the project in Unity, and am faced with the following errors.

`Reference Rewriter found some errors while running with command "--target=Temp\StagingArea\Assembly-CSharp.dll" "--target=Temp\StagingArea\Unity.TextMeshPro.dll" "--target=Temp\StagingArea\Newtonsoft.Json.dll" "--`

Full Error can be found [here](https://gist.github.com/alaksana96/9feed69e7627a8d1f62fdcbebe519be0) (The error is pretty long, I didn't want to copy the whole thing into this issue)

Followed by:

`Reference rewriter: Error: method `System.Xml.XmlWriter System.Xml.XmlWriter::Create(System.String,System.Xml.XmlWriterSettings)` doesn't exist in target framework. It is referenced from Urdf.dll at System.Void RosSharp.Urdf.Robot::WriteToUrdf().
UnityEngine.Debug:LogError(Object)
PostProcessWinRT:RunReferenceRewriter() (at C:/buildslave/unity/build/PlatformDependent/MetroPlayer/Extensions/Managed/PostProcessWinRT.cs:1058)
PostProcessWinRT:Process() (at C:/buildslave/unity/build/PlatformDependent/MetroPlayer/Extensions/Managed/PostProcessWinRT.cs:208)
UnityEngine.GUIUtility:ProcessEvent(Int32, IntPtr)
`
As far as I can tell, I have followed the build instructions properly:

- Unity 2018.2.18
- Visual Studio 2017
- MRTK 2017
As stated on the UWP forks README.md

I'm unfamiliar with the errors above, perhaps I have missed something?

Thanks!
Aufar

---

dwhit, 2019-04-16:
Hmm, I'm not sure what is causing that issue off the top of my head. Probably a new version of the Urdf.dll file is incompatible with the older version of Unity. For now, pull an older version of ros-sharp and use that dll. I'm defending my PhD next week, so I don't have time to make a patch to the UWP branch, but in May I'll be updating both that branch and publishing a new branch with additional topics, and magic leap support.

alaksana96, 2019-04-16:
@dwhit I'll try that.

Didn't mean to distract you from your preparation, best of luck on your thesis defence!

alaksana96, 2019-04-17:
For anyone else facing similar problems, I believe I have partly solved the issue. When I attempt to build the project in Unity, I get the two errors from above followed by a build complete:

[image]

`
Build completed with a result of 'Succeeded'
UnityEngine.GUIUtility:ProcessEvent(Int32, IntPtr)
`

A solution pops up in the App folder, which I can open in VS2017. When I attempt to deploy the program onto the Hololens, the following happens:

[image]

This [Stack Exchange](https://gamedev.stackexchange.com/questions/162445/upgrade-to-unity-2018-2-2f1-resulting-in-missing-file-errors-from-visual-studio) thread solves the issue, and allows me to build and deploy the project to the Hololens. I can then check my rosbridge on the other computer and see a connection. [This](https://answers.unity.com/questions/1529584/cant-use-namespace-tmpro-with-the-textmeshpro-pack.html) sort of explains why the files dont get copied over.

dwhit, 2019-05-17:
Sorry this took so long, but I just updated the UWP branch. It is now compatible with 2019.x, the new MixedRealityToolkit, and the lasted commit of ros-sharp!

alaksana96, 2019-05-19:
Thanks for this!

closed 2019-05-19 by alaksana96