siemens/ros-sharp #195: StandardHeaderExtensions uses time since startup rather than epoch

emnes, 2019-04-18:
* [X] I am at the right place and my issue is directly related to ROS#. General technical questions I would post e.g. at [ROS Answers](https://answers.ros.org/) or [Stack Overflow](https://stackoverflow.com). For library-specific questions I would look for help in the corresponding library forums.
* [X] I have thoroughly read [the Contributing Guideline](Contributing.md) and writing this issue is the right thing to do in my case.

---
## I have a question! ##

* [X] I searched the [Wiki](https://github.com/siemens/ros-sharp/wiki), [open](https://github.com/siemens/ros-sharp/issues) and [closed](https://github.com/siemens/ros-sharp/issues?q=is%3Aissue+is%3Aclosed) issues for an answer. I tried my best to find the answer by myself without success. I believe that the discussion we will have in this issue, and the solutions we might find, will help me, and likely other community members who have a similar problem.

**Here is my question:**

Your implementation of the standard header message type creates a timestamp based on seconds since startup. However, the [std_msgs/Header documentation](https://docs.ros.org/api/std_msgs/html/msg/Header.html) for ROS states that the timestamp should be seconds since epoch

> #Two-integer timestamp that is expressed as:
> # * stamp.sec: seconds (stamp_secs) since epoch (in Python the variable is called 'secs')

I'm interested in the reasoning behind this implementation. Won't this require additional work to keep everything in sync?

MartinBischoff, 2019-04-25:
@emnes There is no big reson behind it. It is simple, the time since the unity simulation epoch and did its job for all our applications so far. If you need to sync between ROS and Unity, you'll have to make adjustments here.

MartinBischoff, 2019-05-10:
@emnes  please have alook at pull request #200 .
It seems to resolve your issue.

MartinBischoff, 2019-05-16:
pull request #200 merged. closing.

closed 2019-05-16 by MartinBischoff