siemens/ros-sharp #196: BSON deserialisation

timstallinger, 2019-04-19:
* [x ] I am at the right place and my issue is directly related to ROS#. General technical questions I would post e.g. at [ROS Answers](https://answers.ros.org/) or [Stack Overflow](https://stackoverflow.com). For library-specific questions I would look for help in the corresponding library forums.
* [ x] I have thoroughly read [the Contributing Guideline](Contributing.md) and writing this issue is the right thing to do in my case.

---
## I have a question! ##

* [ x] I searched the [Wiki](https://github.com/siemens/ros-sharp/wiki), [open](https://github.com/siemens/ros-sharp/issues) and [closed](https://github.com/siemens/ros-sharp/issues?q=is%3Aissue+is%3Aclosed) issues for an answer. I tried my best to find the answer by myself without success. I believe that the discussion we will have in this issue, and the solutions we might find, will help me, and likely other community members who have a similar problem.

**Here is my question:**

Hi everybody,
I have been looking for a way to use BSON with RosSharp to reduce latency. I have read #136 which is marked as a new feature but I don't think it is. When using BSon_only_mode in RosBridge, it says 'Exception in deserialization of BSON. So I have tried to deserialize the recieved Messages in RosSocket:
`MemoryStream ms = new MemoryStream(buffer);`
`using (BsonReader reader = new BsonReader(ms)){`
`    JsonSerializer serializer = new JsonSerializer();`
`    T e = serializer.Deserialize<T>(reader);`
`    return e;`
`}`
I followed this [template](https://www.newtonsoft.com/json/help/html/DeserializeFromBson.htm)
How can I fix the deserialization?
Thank you in advance.

timstallinger, 2019-04-24:
Solved, I had to modify the serialization as well although I did only recieve messages and I didnt send any.

MartinBischoff, 2019-05-09:
Issue has been addressed in commit https://github.com/siemens/ros-sharp/commit/2bcb2666a5af3222da840fd4e87b0a99700fb068 .

closed 2019-05-09 by MartinBischoff