siemens/ros-sharp #201: ROS# UWP TCP Client

tsdavs, 2019-05-13:
* [x] I am at the right place and my issue is directly related to ROS#. General technical questions I would post e.g. at [ROS Answers](https://answers.ros.org/) or [Stack Overflow](https://stackoverflow.com). For library-specific questions I would look for help in the corresponding library forums.
* [x] I have thoroughly read [the Contributing Guideline](Contributing.md) and writing this issue is the right thing to do in my case.

---
## I have a question! ##

* [x] I searched the [Wiki](https://github.com/siemens/ros-sharp/wiki), [open](https://github.com/siemens/ros-sharp/issues) and [closed](https://github.com/siemens/ros-sharp/issues?q=is%3Aissue+is%3Aclosed) issues for an answer. I tried my best to find the answer by myself without success. I believe that the discussion we will have in this issue, and the solutions we might find, will help me, and likely other community members who have a similar problem.

**Here is my question:**
Hi All,

My team and I have successfully built and deployed the @dwhit fork of the ROS# project for the Hololens. Unfortunately, we have run into some issues with socket communication. The project in Unity uses a subscriber class and prints to a UI element. Ultimately the plan is to view this on the Hololens as a kind of HUD element. Currently, the program functions as expected on the Unity IDE but when compiled and built onto the Hololens (or the Hololens sim), **a connection is established to the Rosbridge server but no data is passed through**. We believe that this may be because the issues outlined [here](https://foxypanda.me/tcp-client-in-a-uwp-unity-app-on-hololens/). Should we consider a custom websocket solution like @jaguar243 metioned [here](https://github.com/siemens/ros-sharp/issues/21#issuecomment-346958375)? Does anyone have any advice?

---
* My OS is: Windows 10
* My Unity Version is: 2018.3.0f2
* My ROS Distribution is: Melodic
* My Build Target Platform is: Hololens

Thanks!

Tim

Xend1992, 2019-05-21:
Hi Tim,

I have exactly the same problem as you described. Connection is established, but you can't subscribe to any topic (even when the message type is as simple as standard String). Everything seems to be working fine while testing in the Unity Editor.

Did you come up with any solution to this problem? One of the alternative solutions one could implement could be to use Node-Red with MQTT. There is ros-contrib package you can get for Node-Red which will allow to publish/subscribe (I need to further investigate if it is possible to publish/subscribe topic with complex message types, or just the simple ones). Once the data is captured in Node-Red, I believe it could be broadcasted using MQTT (I previously worked with Hololens and MQTT and the setup is quite efortless, it also work as the opposite to Hololens and ROS).

Regards,
Irek

MartinBischoff, 2019-06-07:
Hi @tsdavs @Xend1992 please make a development build and check the log file for any concrete error messages that we can work with. 
From your reports I did not understand wheter there is no communication because of the incompatiblity of the Websocket Client or because of the  JSON .NET library? For both should be alternatives available that work on your platform.
There are two Websocket Client implementations available in ROS# already. You can add your own by implementing the [IProtocol class](https://github.com/siemens/ros-sharp/blob/master/Libraries/RosBridgeClient/Protocols/IProtocol.cs).
Here is one [JSON .NET implemantion](https://assetstore.unity.com/packages/tools/input-management/json-net-for-unity-11347) to which ROS# is compatible and which might be better suited for our Unity projects than the existing one...

@dwhit do you have any further advice?

MartinBischoff, 2019-06-07:
Please help me with  your platforms guys. Is this issue a duplicate of #184 ?

AsWeSee, 2019-12-18:
Hi @MartinBischoff. I had the exact same issue as described on top and your advice to use [JSON .NET implemantion](https://assetstore.unity.com/packages/tools/input-management/json-net-for-unity-11347) helped.

Short manual to fix the issue:
1) Delete from RosSharp/Plugins Newtonsoft.Json but keep Newtonsoft.Json.Bson
2) Import json asset from the asset store.
There will be errors about possible runtime crashes, but in practice it works fine.

MartinBischoff, 2019-12-19:
Thank your for this info @AsWeSee . It will likely be useful for others having the same problem.

I'm think about replacing replace the official [Newtonsoft JSON .NET](https://www.newtonsoft.com/json) by the [JSON .NET for Unity](https://assetstore.unity.com/packages/tools/input-management/json-net-for-unity-11347) version.

As the [ROS# libraries](https://github.com/siemens/ros-sharp/tree/master/Libraries) can also be used for non-Unity projects, I'd prefer keeping the official version. Also using Newtonsoft.BSON and JSON .NET for Unity in parallel is not so nice, since it seems the latter can handle both JSON and BSON...

berkayalpcakal, 2020-04-21:
Hello @tsdavs, @Xend1992, @AsWeSee,

There has been an improvement in ROS# with #292 which adds a new Json serialization tool. You can get rid of Newtonsoft serialization tool and use .NET Json serialization API instead.  Could you please try and see if it works with UWP? 

I used to have a chance to work with HoloLens, and I successfully setup a proper two way communication bewteen ROS and HoloLens using [this](https://github.com/dwhit/ros-sharp) fork of ROS# (and [this](https://github.com/berkayalpcakal/ROS-HoloLens-Integration) is my repo). Now I do not have the device, so I cannot do the further tests with the new serialization tool.

dnlwbr, 2020-05-19:
Hi @berkayalpcakal, do you mean that #292 could have made the UWP [fork](https://github.com/dwhit/ros-sharp) obsolete? Unfortunately dwith's fork is not quite up to date and was last changed in May 2019. I'm no expert, so I don't know if there have been any important changes since then.

berkayalpcakal, 2020-05-22:
Hi @dnlwbr, I do not think that #292 made UWP fork obsolete. #292 just provides an additional serialization tool so that you can get rid off the Newtonsoft dependency. As far as I remember the biggest modifications in that fork are related to threading related lines of codes, which are required to get ROS# running on a UWP platform. 
 
Even if the UWP fork is old, you can still take advantage of it. I suggest that you clone UWP fork, then search for the use of preprocessor "WINDOWS_UWP" in the entire library solution. You can compare and see what are the major modifications to get ROS# running on a UWP platform.  

I would like to adapt the work in the UWP fork into current ROS#, but I do not have a UWP platform to test it, so it would be nice if you comment your experiences here.

dnlwbr, 2020-06-19:
I can't say much about it yet, but it seems that using Reflection is a problem with UWP builds, see https://github.com/dwhit/ros-sharp/issues/3#issuecomment-634006992. There is a fork that works around this problem (luchspeter/ros-sharp@6583bfd29a103111a6480ac321339015e226c1a2). If you plan to integrate UWP functionality into the current ROS#, you may want to pay attention to this.

helenol, 2020-06-24:
Hi @berkayalpcakal , I couldn't find your e-mail but we're in Zurich and we can loan you a Hololens 1 device. :) (And would love a solution to the UWP problems!) Could you please get in touch with me by e-mail? eloleyni @ microsoft.com

MartinBischoff, 2020-06-24:
Thank you @helenol for the offer to loan a Hololens. 😉 Our problem is not purchasing one. 😉
 
If we decided for officially supporting UWP (or other currently incompatible platforms) here, community would request us to maintain that in the future.
For the UWP, changes are necessary which effectively result in two different ROS# versions. This would effectively double the testing and maintenance effort for all code changes we make from then on.
In case you find a way for us to support UWP without increasing effort or harming the compatibility, performance and ease of use of other platforms, please let us know.
I want to keep ROS# as simple as it is. I hope you understand.

@dwhit kindly provided a ROS# UWP version for some time on his fork.  You, as well as everyone else here is kindly invited to provide platform-specific ROS# forks.

EricVoll, 2020-08-04:
@MartinBischoff 
We performed some minor changes to dwhit's fork and had ROS# running on the HoloLens 1. The maintenance should not be too bad I think. This [commit](https://github.com/EricVoll/ros-sharp/commit/6583bfd29a103111a6480ac321339015e226c1a2) on my fork is all that was needed to run his fork on the HL. The commit also removes some of the compiler-directives that differentiated between UWP/Editor/Windows platform builds etc.

I can help out if you want to.

glennliu, 2020-08-17:
@EricVoll I successfully deployed ROS# on HL by using your fork. 
But there is a problem with message generation. It seems you modified the messages parts. When I use previous sensor_msgs/PointCloud2 generated by ROS#, it cannot be compiled in your fork. And the "rosbridgeclient" message generation is different. Could you please explain how to generate messages with your fork? Thanks
(Sorry for asking a new question. I was trying to ask it in your fork but there isn't an option for "issue")

EricVoll, 2020-08-17:
@glennliu thanks I'll have a look at it today. Should be a quick fix :)

MartinBischoff, 2020-10-12:
Here is the link to @EricVoll UWP fork of ROS#: https://github.com/EricVoll/ros-sharp

closed 2020-10-12 by MartinBischoff