siemens/ros-sharp #208: ROS# WebSocket error handling

Kenichi-Usami, 2019-05-28:
WebSocketNetProtocol.cs and WebSocketSharpProtocol.cs do not have an error handling.
It means IProtocol interface does not include the error event handler.

Could you add  the event handler such as onError if it is useful?

I would like to get any error by the error event handler when disconnect and so on.
I understand the error event handler can not work perfectly because of network situation is complicated.
But if it has some advantage, I want the handler.

giorgiovaccarino, 2019-05-28:
I agree with Kenichi-Usami. I use ros-sharp in a ASP.NET Core application and if a socket error arise the entire web application process crash.
For this I modified in a rough way ConnectAsync, SendAsync, StartListen.
It is far from perfect, I haven't had time to review the code yet (i.e. the connection error don't emerge to callers) but meanwhile it prevents the application from crash.

        public Exception ConnectException { get; private set; }

        public async void ConnectAsync()
        {
            try
            {
                await clientWebSocket.ConnectAsync(uri, cancellationToken);
                IsConnected.Set();
                OnConnected?.Invoke(null, EventArgs.Empty);
                StartListen();
            }
            catch (Exception exception)
            {
                clientWebSocket.Abort();
                ConnectException = exception;
            }
        }

        public async void SendAsync(byte[] message)
        {
            IsConnected.WaitOne();

            //if (clientWebSocket.State != WebSocketState.Open)
            //    throw new WebSocketException(WebSocketError.InvalidState, "Error Sending Message. WebSocket State is: " + clientWebSocket.State);

            if (clientWebSocket.State == WebSocketState.Open)
            {
                int messageCount = (int)Math.Ceiling((double)message.Length / SendChunkSize);

                IsReadyToSend.WaitOne();
                for (int i = 0; i < messageCount; i++)
                {
                    int offset = SendChunkSize * i;
                    bool endOfMessage = (i == messageCount - 1);
                    int count = endOfMessage ? message.Length - offset : SendChunkSize;
                    await clientWebSocket.SendAsync(new ArraySegment<byte>(message, offset, count), WebSocketMessageType.Text, endOfMessage, cancellationToken);
                }
                IsReadyToSend.Set();
            }
        }

        private async void StartListen()
        {
            byte[] buffer = new byte[ReceiveChunkSize];
            while (clientWebSocket.State == WebSocketState.Open)
            {
                MemoryStream memoryStream = new MemoryStream();
                WebSocketReceiveResult result;
                do
                {
                    try
                    {
                        result = await clientWebSocket.ReceiveAsync(new ArraySegment<byte>(buffer), cancellationToken);
                        if (result.MessageType == WebSocketMessageType.Close)
                            return;
                        memoryStream.Write(buffer, 0, result.Count);
                    }
                    catch
                    {
                        return;
                        //break;
                    }
                } while (!result.EndOfMessage);
                OnReceive?.Invoke(this, new MessageEventArgs(memoryStream.ToArray()));
            }
        }

MartinBischoff, 2019-05-28:
Sounds good. We'll accept a pull request for error handling. It should consist of an extended IProtocol interface and the corresponding implementations for WebSocketNetProtocol.cs and WebSocketSharpProtocol.cs.

MartinBischoff, 2020-04-21:
done in the commit above long ago.

closed 2020-04-21 by MartinBischoff