siemens/ros-sharp #210: Help! Can't not access the variable linearVelocity of TwistPublisher.cs from another script in Unity

stevensu1838, 2019-06-02:
Hi all,
My OS is: Windows 10
My Unity Version is: Unity 2018.4
My ROS Distribution is: Kinetic
My ROS # Version is:1.4

I am trying to read the variable linearVelocity of TwistPublisher.cs from another script which is attached on a normal 3D gameobject in Unity. 
As you know,  linearVelocity of TwistPublisher.cs can represent the velocity of a related gameobject in unity(In my case, the related gameobject is a cube), I simply want to access this linearVelocity from another script which I named CsvWriteScript.
Please take a look at my CsvWriteScript:

```
using System.Collections;
using System.Collections.Generic;
using UnityEngine;
using System.Text;
using System.IO;
using System;
using RosSharp.RosBridgeClient;

public class CsvWriteScript : MonoBehaviour
{
    GameObject obj;
    TwistPublisher twistpublisherScript;

    void Start()
    {  
        obj = GameObject.FindGameObjectWithTag("RosConnector");
        twistpublisherScript = obj.GetComponent<TwistPublisher>();
    }

    void Update()
    {
        UnityEngine.Debug.Log("the velocity of this cube:" + twistpublisherScript.linearVelocity);
    }
}

```
In your TwistPublisher.cs  I used  Debug.Log("Vlinear" + linearVelocity); to print the velocity
```
void UpdateMessage()
        {
            float deltaTime = Time.realtimeSinceStartup - previousRealTime;
            Vector3 linearVelocity = (PublishedTransform.position - previousPosition) / deltaTime;
            Vector3 angularVelocity = (PublishedTransform.rotation.eulerAngles - previousRotation.eulerAngles)/deltaTime; 
            //Steven
            Debug.Log("Vlinear" + linearVelocity);
```
 Then I moved the cube in unity by dragging it with the mouse, it turned out the print from Debug.Log("Vlinear" + linearVelocity); was updating but the print from UnityEngine.Debug.Log("the velocity of this cube:" + twistpublisherScript.linearVelocity); was always(0, 0 , 0)   

I am sure the link between your TwistPublisher.cs and my CsvWriteScript is built. Any chance you know why the twistpublisherScript.linearVelocity does't update when the linearVelocity from your TwistPublisher.cs is changing? 

Thanks a million for your help.

To find out the problems, I write my own script to read the velocity of the cube object to replace your TwistPublisher.cs. The script is shown below 

```
using System.Collections;
using System.Collections.Generic;
using UnityEngine;

public class ObjVelocityReader : MonoBehaviour
{
    Vector3 PrevPos;
    Vector3 NewPos;
    Vector3 PrevRot;
    Vector3 NewRot;

    public Vector3 ObjVelocity;
    public Vector3 ObjRotation;
    // Use this for initialization
    void Start()
    {
        PrevPos = transform.position;
        PrevRot = transform.eulerAngles;
    }

    // Update is called once per frame
    void FixedUpdate()
    {
        transform.rotation = Quaternion.identity;

        NewPos = transform.position;  // each frame track the new position
        ObjVelocity = (NewPos - PrevPos) / Time.fixedDeltaTime;
        PrevPos = NewPos;  // update position for next frame calculation

        NewRot = transform.eulerAngles;  // each frame track the new rotation
        ObjRotation = (NewRot - PrevRot) / Time.fixedDeltaTime;
        PrevRot = NewRot;  // update position for next frame calculation
    }
}
      

```

And then I used my  CsvWriteScript(shown below)to access the velocity variable from my ObjVelocityReader（it is a replacement to your TwistPublisher.cs here), it was a success. I could access the velocity from my CsvWriteScript successfully. So I really couldn't figure out how I should modify your TwistPublisher.cs to fix the problem. Can you please do me a favor? Really appreciated.

```
using System.Collections;
using System.Collections.Generic;
using UnityEngine;
using System.Text;
using System.IO;
using System;
using RosSharp.RosBridgeClient;

public class CsvWriteScript : MonoBehaviour
{
    GameObject obj;
    ObjVelocityReader velocityScript;

    // Start is called before the first frame update
    void Start()
    {
        //obj = GameObject.FindGameObjectWithTag("RosConnector");
        //twistpublisherScript = obj.GetComponent<TwistPublisher>();
        obj = GameObject.Find("Sphere");
        velocityScript = obj.GetComponent<ObjVelocityReader>();        
    }
    void Update()
    {
        UnityEngine.Debug.Log("V:" + velocityScript. ObjVelocity);
    }
}

```

stevensu1838, 2019-06-03:
Problem solved by removing Vector3 in front for linearVelcity in TwistPubliser.cs. Not sure why.
```
Vector3 linearVelocity = (transform.position - previousPosition)/deltaTime;
Vector3 angularVelocity = (transform.rotation.eulerAngles -previousRotation.eulerAngles)/deltaTime;
```

Now it is:
```
namespace RosSharp.RosBridgeClient
{
    public class TwistProvider : MessageProvider
    {
        public Vector3 linearVelocity;
        

        private void UpdateMessage()
        {
        Vector3 angularVelocity = (transform.rotation.eulerAngles -previousRotation.eulerAngles)/deltaTime;
         }
```

MartinBischoff, 2019-06-07:
@stevensu1838 Please mind the issue template. Please address this issue board only for questions directly related to ROS#. For general programming questions there are [better information sources available](http://lmgtfy.com/?q=c%23+variable+scope).
Closing #210 .

closed 2019-06-07 by MartinBischoff