siemens/ros-sharp #214: Can't build RosSharp.sln in Visual Studio successfully

stevensu1838, 2019-06-12:
Hi all,

First, I'd like to say thank you to Martin and all of you who have developed this package. It has been super useful.

Secondly, please let me explain how I meet these errors.
1. down load ros-sharp-master to my D: from https://github.com/siemens/ros-sharp.git
2. Unzip the package and then go to directory D:\Steven\Unityprojects\ros-sharp-master\ros-sharp-master\Libraries\RosBridgeClient\Messages\Sensor to create a script named Range.cs
3.  Check closely the ROS sensor msg/range  document 

sensor_msgs/Range Message
File: sensor_msgs/Range.msg
Raw Message Definition

```
Header header           # timestamp in the header is the time the ranger
                                   # returned the distance reading
# Radiation type enums
# If you want a value added to this list, send an email to the ros-users list
uint8 ULTRASOUND=0
uint8 INFRARED=1
uint8 radiation_type    # the type of radiation used by the sensor
float32 field_of_view   # the size of the arc that the distance reading is
float32 min_range       # minimum range value [m]
float32 max_range       # maximum range value [m]
float32 range           # range data [m]
                    
Compact Message Definition
uint8 ULTRASOUND=0
uint8 INFRARED=1
std_msgs/Header header
uint8 radiation_type
float32 field_of_view
float32 min_range
float32 max_range
float32 range

```
4. Fill my range.cs file as shown below:
```
using Newtonsoft.Json;
namespace RosSharp.RosBridgeClient.Messages.Sensor
{
    public class Range : Message
    {
        [JsonIgnore]
        public const string RosMessageName = "sensor_msgs/Range";
        public Standard.Header header;
        public const byte ULTRASOUND = 0;
        public const byte INFRARED=1;
        public uint radiation_type;
        public float field_of_view;
        public float min_range;
        public float max_range;
        public float range;

        public Range()
        {
            header = new Standard.Header();
            radiation_type = 0;
            field_of_view = 0;
            min_range = 0;
            max_range = 0;
            range_min = 0;
            range_max = 0;
            range = 0 ;
        }
    }
}

```
Can any of you experts take a look as my range.cs above? I am very confused on uint8 ULTRASOUND=0 and uint8 INFRARED=1. I am not sure I should use  public const byte or public uint type for them in my range.cs.

May I please post the errors it throws in Visual Studio when building RosSharp.sln in Visual Studio to save your time? Could you please consider as range.cs into you official ros-sharp-master?

stevensu1838, 2019-06-13:
Hi all,
When I move on to Build RosSharp.sln in Visual Studio, I always see the following errors. Any help?

Build RosSharp.sln in Visual Studio
[image]

stevensu1838, 2019-06-13:
Hi all,
Just want to share the solution to fix error: .NETFramework,Version=v4.6.2” were not found. I reckon I won't be the only person facing it.

1. Please downland and install Microsoft .NET Framework 4.6. Developer download. Make sure it is a Developer Pack.  
https://www.microsoft.com/en-nz/download/details.aspx?id=49978
2. Restart visual studio after installing.
Errors solved.

But can anyone tell me how shall I convert the following component into my range.cs? It's been very confusing. Thanks
uint8 ULTRASOUND=0
uint8 INFRARED=1
uint8 radiation_type    # the type of radiation

berkayalpcakal, 2019-06-14:
Hello @stevensu1838 

I am sorry that I did not get why you use const byte there. The documentation of the message says that those message members should have the data type "uint8". The word "enum" must have been confusing for you, but numeric constants are mostly referred as enums on ROS message structures. So you should use a data type in C# which is equivalent to uint8.

[This link should help!](http://lmgtfy.com/?q=uint8+in+c%23)

closed 2019-06-14 by berkayalpcakal

stevensu1838, 2019-06-17:
Hi 
May I please first explain why I used const byte there in my Range.cs? It was totally because I was using ros sensor_msg/Pointfeild and your Pointfeild.cs as examples.  And the reason why I take ros sensor_msg/Pointfeild was I saw the following component in both ros sensor_msg/pointfeild and sensor_msg/Range(Range is the one I've been working on.)
[image]
Can you please tell me why const byte is used above?
I've noticed you normally use public byte in your code to represent the data type "uint8

stevensu1838, 2019-06-17:
Hi, May I please show you my Range.cs and Rangsubscriber.cs and the errors that I have in Unity?

```
using Newtonsoft.Json;

namespace RosSharp.RosBridgeClient.Messages.Sensor
{
    public class Range : Message
    {
        [JsonIgnore]
        public const string RosMessageName = "sensor_msgs/Range";
        public Standard.Header header;
        public byte ULTRASOUND = 0;
        public byte INFRARED = 1;
        public byte radiation_type;
        public float field_of_view;
        public float min_range;
        public float max_range;
        public float range;

        public Range()
        {
            header = new Standard.Header();
            radiation_type = 0;
            field_of_view = 0;
            min_range = 0;
            max_range = 0;
            range_min = 0;
            range_max = 0;
            range = 0;
        }
    }
}
```

namespace RosSharp.RosBridgeClient
{
    public class RangeSubscriber : Subscriber<Messages.Sensor.Range>
    {    
        //public LaserScanWriter laserScanWriter;
        //public Transform SubscribedTransform;

        //private float previousRealTime;
        //private Vector3 linearVelocity;
        //private Vector3 angularVelocity;
        private bool isMessageReceived;

        public float rangeData;
        public float range_min;
        public float range_max;

        protected override void Start()
        {
            base.Start();
        }

        private void Update()
        {
            if (isMessageReceived)
                ProcessMessage();
        }

        protected override void ReceiveMessage(Messages.Sensor.Range range)
        {
            //laserScanWriter.Write(laserScan);

            rangeData = range.range;
            range_min = range.min_range;
            range_max = range.max_range;
            Debug.Log("Hello: " + rangeData);
            isMessageReceived = true;
        }
        private void ProcessMessage()
        {
            
        }

    }
}

[image]

Please help