siemens/ros-sharp #217: PointCloud2 lag.

bomarali, 2019-06-17:
I am at the right place and my issue is directly related to ROS#. General technical questions I would post e.g. at [ROS Answers](https://answers.ros.org/) or [Stack Overflow](https://stackoverflow.com). For library-specific questions I would look for help in the corresponding library forums.
I have thoroughly read [the Contributing Guideline](Contributing.md) and writing this issue is the right thing to do in my case.

---
## I have a question! ##

I searched the [Wiki](https://github.com/siemens/ros-sharp/wiki), [open](https://github.com/siemens/ros-sharp/issues) and [closed](https://github.com/siemens/ros-sharp/issues?q=is%3Aissue+is%3Aclosed) issues for an answer. I tried my best to find the answer by myself without success. I believe that the discussion we will have in this issue, and the solutions we might find, will help me, and likely other community members who have a similar problem.

**Here is my question:**
I am streaming a PointCloud2 from a ROS to Unity using a local LAN. The server publishes the pcl2 at 30Hz at sd quality (kinect2) but the Unity reads it at approximately 2Hz. There seems to be a communication bottleneck. 
I have a custom udp connection setup that manages to stream the pcl2 loseless. So I assume it is not a network issue but json or rosbridge issue? There is no considerable difference in frequencies when using the PoseStamped message.
Could you tell me if there are communication restrictions in ros#?

bomarali, 2019-06-17:
The lag disappears when the pcl2 size is reduced from from sd (524 by 424 or something) to 16000 points (downsampled it agressively with a pcl_nodelet). This tells me that either server doesnt push the data to the socket fast enough or the client fails to read it fast enough. 
Next I have throttled down the subscriber in unity on full sd pcl2 at 1s. The lag was gone but the resulting visualisation suffers greatly. Assuming that the server is not aware of throttle rate (is it?) I conclude that the server does publish on time, it is the client that can't read data fast enough. 
I am aware that in the kinect2 half of the data payload is actually empty so I could get some performance boost by removing the non-essential bits. 
Yet I am more interested in getting all of the data from kinect2 to unity lossless. Any other ideas how to speed it up?

bomarali, 2019-06-17:
I have tried to run in BSON, no effect. I also changed compression to "png" but then the point cloud would not be read at all.

closed 2019-10-28 by bomarali

HMellor, 2021-02-17:
Hi @bomarali, I have noticed the same bottleneck type behaviour when trying to visualise a `PointCloud2` topic from ROS. Were you able to figure out what was causing it?

bomarali, 2021-02-18:
I think it might have something to do with the size of the point cloud. In my workaround I simply pushed the point cloud through a separate dedicated UDP channel. I is mentioned here: https://ieeexplore.ieee.org/abstract/document/9223445?casa_token=Gg7ce_XqrqEAAAAA:hEhIUZuRQXWw3aa-ZbrPBYMchuT6Mpz3xNmXHPUhCBm3J9EyOqUKsqshcVCXkPtmrgmeHmHZ-A

MartinBischoff, 2021-02-18:
Hi @HMellor ,
The problem is JSON / BSON serialization which is bad for image data.
For this reason Unity Robotics Hub setup a simple direct tpc connection bypassing rosbridge_suite.
See this issue on rosbridge_suite for more info: https://github.com/RobotWebTools/rosbridge_suite/issues/546
As mentioned [here](https://github.com/siemens/ros-sharp/wiki/Ext_RosSharp_RoboticsHub), we're interested in adding further IProtocol implementations which are better suited for image data.