siemens/ros-sharp #219: ROS connection text field

BourdonValentin, 2019-06-18:
Hi, 

I would like to create a connection by passing directly an IP address (ws://xxx.xxx.xxx:9090) with an input filed text on my game view to modify it when I want and when is necessary. 
I tried this solution but the connection always failed or the URI is invalid. However, when I pass the IP address directly in my script or when I modified it in the inspector of the RosConnector script it works ! 

Have you a solution for this ?

Thanks !

bomarali, 2019-06-18:
as far as i remember there is a lot of rosconnection that happens on start() or on awake(). These calls will happen long before you use the input field. So my guess is you will need to modify everything that work on start and awake to work from a button prompt or something. 
Alternatively may be you can use multiple unity levels (scenes). You can have an entry level where you input connection specifics that needs to be set as persistent when loading next level that is the generic ros# level.

BourdonValentin, 2019-06-18:
Thanks you very much ! I will  try this.

closed 2019-06-18 by BourdonValentin