siemens/ros-sharp #221: I can not send and receive simple data with ROS #

JuanCarlosGallo, 2019-06-22:
## Hello ##

I am new using ROS # and I am not very expert using ROS, but I hope to receive your valuable help. I am trying to send data from UNITY3D to ROS using ROS #, I do not need to send or import a URDF because they are simple data like a String, INT or FLOAT.

I have already done the configuration steps and the examples of the wiki https://github.com/siemens/ros-sharp/wiki (Sections 1, 2 and 3). I built a new project in UNITY3D, I create an empty GameObject called "messenger" and add the RosConnector script, select JSON and web socket net and place the IP 192.168.56.101:9090 of the virtual machine.

[image]

Then create a new script called ConsoleExample and place the same code as the wiki example https://github.com/siemens/ros-sharp/wiki/Info_CodeExample to start testing but some problems appeared:

1) there were two compilation errors with Protocols and std_msgs.String.

[image]

the first one solved it by adding the name space RosSharp.RosBridgeClient, but I could not solve the second so I commented on the publication section to test only the message subcription, there were no more errors but I would like help to solve the second error and test the publication of messages.

2) I tried adding the Script ConsoleExample to the mesennger GameObject, but a message appeared saying that ConsoleExample needs to be derived from MonoBehaviour, then add: MonoBehaviour and using UnityEngine; to the code, the code stayed that way.

[image]

Now add the Script ConsoleExample to the GameObject mesennger

[image]

So far 
* I do not know if I've done something wrong
* I do not know if I've built the project well
* why do those errors appear? 
* Are the lines I add correct?
* How can I try the message publishing section?

## Now try with ROS ##

In the virtual machine run ubuntu and ros, in a terminal only execute the following commands:
* `roslaunch rosbridge_server rosbridge_websocket.launch`
* `rostopic pub /subscription_test std_msgs/String "subscription test message data"`

[image]

Play the application in UNITY3D the connection with ROS is well established, but I do not know where the message is received?

[image]

 I have only found these examples:
* https://github.com/siemens/ros-sharp/wiki/Info_CodeExample
* https://github.com/siemens/ros-sharp/blob/master/Libraries/RosBridgeClientTest/RosSocketConsoleExample.cs
* https://rosindustrial.org/news/2018/1/8/announcing-ros
* https://github.com/siemens/ros-sharp/wiki/Dev_NewMessageTypes

But I would like to know if someone has a manual where I can find more detailed information about how to send and receive messages with ROS #

I am used:
* Windows 10
* Unity 2018.3.14
* VirtualBox 6.0
* Ubuntu 16.04
* Ros Kinetic 

Thanks for your help

JuanCarlosGallo, 2019-06-28:
Hi @bgonzupenn I'm glad to know that I'm not the only one in the world with this problem

I was working and found a solution based on the examples of the wiki Info_[Info_CodeExample](https://github.com/siemens/ros-sharp/wiki/Info_CodeExample) and [ RosSocketConsoleExample](https://github.com/siemens/ros-sharp/blob/master/Libraries/RosBridgeClientTest/RosSocketConsoleExample.cs). I could see that the `public static void Main (string [] args)` function of Info_CodeExample was never executed, I DO NOT KNOW WHY. So I decided to work with the functions used by default UNITY `void Start()` and `void Update()`, my Script is shown below: [ExampleSendReceive.zip](https://github.com/siemens/ros-sharp/files/3340237/ExampleSendReceive.zip)

```using System;
using System.Collections;
using System.Collections.Generic;
using RosSharp.RosBridgeClient;
using std_msgs = RosSharp.RosBridgeClient.Messages.Standard;
using UnityEngine;

namespace RosSharp.RosBridgeClient
{

    public class ExampleSendReceive : MonoBehaviour
    {
        RosSocket rosSocket;
        string publication_id;
        string subscription_id;

        // Start is called before the first frame update, here the connection with ROS is established and the tipics are defined
        void Start()
        {
            rosSocket = new RosSocket(new RosBridgeClient.Protocols.WebSocketNetProtocol("ws://192.168.56.101:9090")); //this line does the same as the script RosConnector
            Debug.Log("Established connection with ros");

            publication_id = rosSocket.Advertise<std_msgs.String>("publication_test");

            subscription_id = rosSocket.Subscribe<std_msgs.String>("/subscription_test", SubscriptionHandler);
        }

        // Update is called once per frame
        void Update()
        {
            if (Input.GetKeyDown(KeyCode.S))            //press S key every time you want to send messages to ROS
            {                
                std_msgs.String message = new std_msgs.String
                {
                    data = "Message sent from unity"    //UNITY message for ROS
                };
                rosSocket.Publish(publication_id, message);
                Debug.Log("A message was sent to ROS");
            }

            if (Input.GetKeyDown(KeyCode.C))            //press C key to close the connection
            {
                rosSocket.Close();
                Debug.Log("Closed connection");
            }
        }

        private static void SubscriptionHandler(std_msgs.String message)
        {
            Debug.Log(message.data);                    //print ROS's message to unity
        }
    }
}
```
I also noticed that the line `rosSocket = new RosSocket(new RosBridgeClient.Protocols.WebSocketNetProtocol("ws://192.168.56.101:9090"));` does the same thing as the RosConnector script and that's why I removed it from the project. In this way the project in UNITY remained this way:

[image]

To execute, you must do the following:

- [ ] In a UBUNTU terminal, run `roslaunch rosbridge_server rosbridge_websocket.launch`

- [ ] Play to the application of UNITY

- [ ] In another terminal of UBUNTU run  `rostopic echo /publication_test` to be able to see the message

- [ ] To send a message to ROS from UNITY, press the S key (verify that you have selected the GAME screen) the following will appear

- [ ] To send a message to UNITY from ROS, execute `rostopic pub /subscription_test std_msgs/String "subscription test message data"` in a new UBUNTU terminal, the following will appear

bgonzupenn, 2019-06-28:
Hi @JuanCarlosGallo - this is really great!! Thanks so much for the advice - I found that the [example code](https://github.com/siemens/ros-sharp/wiki/Info_CodeExample) that you mentioned for communication was also not really functional for me.

Like you stated, it has a **main()** function - I think this is because the code is written as a general ROS# example, and not specifically for Unity. Thus, it doesn't account for the fact that Unity must use the **Start()/Update()** functions for its scripts.

I used your script and everything works smoothly to send messages from ROS to Unity, which is what I need. I'd definitely encourage the Wiki to put up your example script as a guide for users who are specifically looking for simple communication between ROS and Unity. Thanks again! 😁

PakoYam, 2019-07-31:
Hi @JuanCarlosGallo I have tried your codes and they work.
I am totally new to ROS and I am wondering how to write a node in ROS to subscribe the topic "publication_test" and use "ROS_INFO" to print it out. 
Thanks!

closed 2019-09-17 by berkayalpcakal