siemens/ros-sharp #224: ROS# to connect to real robots

skouicy, 2019-07-04:
Hi Martin, good job on the ros sharp and the examples you are providing. I have managed to get everything working properly and i found your tutorials very easy to understand and accurate to follow. 
I was wondering if you would be interested to provide some additional information as to how to actually connect a real life robot and teleoperate using unity, or perhaps make a tutorial for us about it. Lets say a generic ROS supported robot like NAO or Pioneer 3AT.
How easy is to actually move away from the gazeebo simulation to the real world? Is is as simple as installing rosbridge libraries you are providing in the tutorial pages and then roslaunch rosbridge_server rosbridge_websocket.launch?
I believe that a general tutorial demonstrating how to use your library for other robots and in reality would be very beneficial for us robot enthousiasts.
Thank you again for your hard work

MartinBischoff, 2019-07-05:
Hi @skouicy :

>  Is is as simple as installing rosbridge libraries you are providing in the tutorial pages and then roslaunch rosbridge_server rosbridge_websocket.launch?

Yes!
Looking at the interfaces from the Unity / Windows side, there is no difference between communicating with a virtual or real robot. We are communicating with a ROS system via rosbridge_suite and Websocket connection. For ROS# I see nothing new to explain or document here. 
The nice thing about the virtual Turtlebot 2 example is that everybody can get it running with nothing but one PC and free software. 
For simplicity / learnability / maintainability we should focus on explaining only things relevant to ROS#, and as little as possible about connected systems (other robots, ROS, Unity, Visual Studio, C#, JSON, VR/AR Glasses, Rendering, ...). For those topics there is (and will be) better documentation around elsewere already.
Have fun!

skouicy, 2019-07-07:
Hi @MartinBischoff and thanks for your reply.
I can understand why you dont want to get into the discussion of introducing instructions for connecting robots etc, as this will shift away the point of this wiki.

However, considering that this framework may be sought by people who are perhaps interested in doing such projects, would it be possible to at least provide more information how you have connected the turtlebot 2 as per your youtube video: https://www.youtube.com/watch?v=OytzagQirrk&feature=youtu.be

For instance, as there is no need for a gazeebo implementation etc as indicated on your wiki site, what is the configuration and commands you have issued to the turtlebox in order to get it working? 
i.e. install rosbridge server on the robot: $ sudo apt-get install ros-kinetic-rosbridge-server
Do we need the joy_to_twist script? and if so where to place it etc. 

I believe that this would be of great benefit to everyone, as we can study your example and configure it accordingly to other robots.
Looking forward to your reply.

closed 2019-07-08 by MartinBischoff