siemens/ros-sharp #227: WebsocketClosedError : Tried to write to a closed websocket

TomSevestre, 2019-07-15:
**Here is my question:**
I'm trying to subscribe to a camera on a robot with the Image Subscriber Script, but I have a problem, the robot keep sending me this : 
"WebSocketClosedError : Tried to write to a closed websocket"
And I don't know what it means (I'm new to use ROS and Unity)
How can I fix this warning and receive correctly the image of the camera ?

---

gopikaajaykumar, 2019-07-22:
I have the same issue. I believe the error first appeared when I started using the RosSocket that I had initially only used for subscribing to the joint states of a UR5 robot to also publish the poses of some game objects and to send and receive messages for an action client. I am not explicitly closing any sockets in my code (not counting the code that closes the socket in RosConnector script).  I have not been able to figure out the problem yet, so any suggestions would help.

gopikaajaykumar, 2019-07-23:
Just a note that although I am getting this warning, I am still able to successfully publish messages/send goals to an action server from Unity. I created new RosSockets for the action client and for publishing Pose messages. Not sure if that is appropriate since I don't know too much about socket programming, but it's working for me, regardless of the warning message. Unfortunately, I haven't been able to figure out what is causing the WebSocketClosedError, though.

closed 2019-07-24 by TomSevestre

Brabalawuka, 2019-08-01:
Why you closed this, have u figured out a fix?

bomarali, 2019-08-27:
Sorry for necromancing but is the source of the error have been found? 
I believe I have a similar thing happening to my setup. The Unity client connects to the server and then I get WebSocketClosedError every second. After client disconnects I get the Could not process inbound connection error.
This behavior started about a month ago, before that everything was clean. I have updated the ros-sharp to the latest master.